introlab/rtabmap #1048: rtabmap/cloud_map is empty

den-schmidt, 2023-06-07:
Hi Matthieu,

we are trying to build a self-driving car and would like to use rtabmap. Our setup:

- Jetson
- Docker container with
        - Ubuntu 18.04
        - ROS Melodic
- we installed rtabmap using
        - sudo apt-get update
        - sudo apt install ros-melodic-rtabmap-ros
        - everything worked
- RealSense d435i with realsense-ros package (apt install too)

We follow the tutorial on "Kinect-like handheld mapping" using the code for the d435i (imu filter installed too) and everything seems to run. Rtabmap updates with a frequency of 1Hz it says. But the topic rtabmap/cloud_map is only published once at the beginning and then it is empty even though we drive the car slowly with the remote control. 

Also, following the tutorial "SLAM with d435i" on the Wiki page of realsense-ros using the opensource_tracking.launch brings the same problem.

Could you please share some ideas on how to debug rtabmap with us? We would be really happy! :)

matlabbe, 2023-06-11:
Are there warnings/errors in the terminal? Maybe the odometry gets lost?

For a general recommendation with D435i, use the IR+Depth or Stereo IR approach. See this tutorial: http://wiki.ros.org/rtabmap_ros/TutorialsOldInterface/StereoHandHeldMapping

den-schmidt, 2023-06-11:
Hi @matlabbe ,

thank you for your response. 

Yes, we noticed that after 3 seconds of moving slowly with the car, the odom frame just grays out and disappears in rviz.

den-schmidt, 2023-06-11:
(1) I have seen that alternative on your website and will try it out tomorrow but I did not understand the difference between both approaches, IR+Depth or Stereo IR. Could you maybe explain them in one sentence?

(2) Do you think we could use RTAB-Map with a D435i on a 1/8 race car with only white lane markings and a black floor? From the Kinect challenge in 2014, I have seen that one can adjust the region of interest which in our case would not be the upper 70% of the image but the lower 30%.

Thank you in advance! :)

closed 2023-06-11 by den-schmidt

matlabbe, 2023-06-11:
(1) 
* IR-Depth means: Left IR camera + Depth computed on-device (which is already aligned with left IR camera), used as RGB-D camera from rtabmap point of view.
* Stereo IR: Left IR + Right IR cameras, used as stereo camera from rtabmap point of view, disparity is computed on host.

On both cases, they are better than RGB-D mode of realsense because the IR cameras have large field-of-view than the RGB camera, they also have less motion blur, and synchronization of the IR topics are perfect.

(2) Lanes on the floor, if there are no other visual features, may not be the best for visual odometry. However, if the upper part of the image has plenty of visual features to track, that could work. The kinect challenge was unstructured environment, but in your case, you may be able to have a local odometry approach based on lane detection (that could be faster and more accurate than visual odometry).