introlab/rtabmap #1049: Is it possible to use rtabmap with FRAMOS D435E?

tall1, 2023-06-08:
The FRAMOS camera is a packed camera based on RealSense D435i.

I'm wondering if I can use rtabmap with this camera.
It publishes all the topics needed to run in D435i but when I follow this [tutorial](http://wiki.ros.org/rtabmap_ros/Tutorials/HandHeldMapping#Mapping_mode), I can't seem to get an Image and all I'm Getting is:

`[ WARN] [1686215747.498701814] [/rtabmap/rtabmap_viz]: /rtabmap/rtabmap_viz: Did not receive data since 5 seconds! Make sure the input topics are published ("$ rostopic hz my_topic") and the timestamps in their header are set. If topics are coming from different computers, make sure the clocks of the computers are synchronized ("ntpdate"). If topics are not published at the same rate, you could increase "queue_size" parameter (current=10).
/rtabmap/rtabmap_viz subscribed to (approx sync):
   /rtabmap/odom \
   /camera/color/image_raw \
   /rtabmap/camera/depth/color/points \
   /camera/color/camera_info \
   /rtabmap/odom_info
`

The launch command I use for rtabmap:
```
roslaunch rtabmap_launch rtabmap.launch \
    rtabmap_args:="--delete_db_on_start --Optimizer/GravitySigma 0.3" \
    depth_topic:=camera/depth/color/points \
    rgb_topic:=/camera/color/image_raw \
    camera_info_topic:=/camera/color/camera_info \
    approx_sync:=false \
    wait_imu_to_init:=true \
    imu_topic:=/rtabmap/imu
```

And the launch file itself (rtabmap.launch):
```
<launch>
    <!-- 
          NOTICE: Installation of rtabmap is required for using this launch file:
                  For installation type:
                        apt-get install ros-kinetic-rtabmap-ros
    -->
    <arg name="device_type_camera1"    		default="t265"/>
    <arg name="device_type_camera2"    		default="d4.5"/>	<!-- Note: using regular expression. match D435, D435i, D415... -->
    <arg name="serial_no_camera1"    			default=""/>
    <arg name="serial_no_camera2"    			default=""/>
    <arg name="camera1"              			default="t265"/>		<!-- Note: Replace with camera name -->
    <arg name="camera2"              			default="d400"/>		<!-- Note: Replace with camera name -->
    <arg name="clip_distance"             default="-2"/>
    <arg name="use_rviz"                  default="true"/>
    <arg name="use_rtabmapviz"            default="false"/>
    

    <include file="$(find realsense2_camera)/launch/rs_d400_and_t265.launch">
            <arg name="device_type_camera1"             value="$(arg device_type_camera1)"/>
            <arg name="device_type_camera2"             value="$(arg device_type_camera2)"/>
            <arg name="serial_no_camera1"               value="$(arg serial_no_camera1)"/>
            <arg name="serial_no_camera2"               value="$(arg serial_no_camera2)"/>
            <arg name="camera1"                         value="$(arg camera1)"/>
            <arg name="camera2"                         value="$(arg camera2)"/>
            <arg name="clip_distance"                   value="$(arg clip_distance)"/>
            
    </include>

    <include file="$(find rtabmap_ros)/launch/rtabmap.launch">
            <arg name="rtabmap_args"       value="--delete_db_on_start"/>
            <arg name="depth_topic"        value="/$(arg camera2)/aligned_depth_to_color/image_raw"/>
            <arg name="frame_id"           value="$(arg camera2)_link"/>
            <arg name="visual_odometry"    value="false"/>
            <arg name="odom_topic"         value="/$(arg camera1)/odom/sample"/>
            <arg name="rgb_topic"          value="/$(arg camera2)/color/image_raw"/>
            <arg name="camera_info_topic"  value="/$(arg camera2)/color/camera_info"/>
            <arg name="queue_size"         value="200"/>
            <arg name="rviz"               value="$(arg use_rviz)"/>
            <arg name="rtabmapviz"         value="$(arg use_rtabmapviz)"/>
    </include>
</launch>
```

matlabbe, 2023-06-11:
Hi,

you are mixing `rs_d400_and_t265.lanuch` (which would be for 2 cameras) and another `rtabmap.launch` not remapping the same arguments that you are passing in command line. 

If you start only the FRAMOS camera, which topics are available? (`rostopic list`)

tall1, 2023-06-15:
I'm currently working on some other more urgent things.

Will update when I get to this.