introlab/rtabmap #1051: iOS build failed

patrickwasp, 2023-06-09:
I installed the prerequisites listed and ran the commands to prepare the environment to compile for iOS and get the error shown below. I also tried downloading and preparing the latest release (https://github.com/introlab/rtabmap/archive/refs/tags/0.21.0.zip) with the same effect. 

`prerequisites`
```
git version 2.39.2 (Apple Git-143)

curl 7.86.0 (x86_64-apple-darwin22.0) libcurl/7.86.0 (SecureTransport) LibreSSL/3.3.6 zlib/1.2.11 nghttp2/1.47.0
Release-Date: 2022-10-26
Protocols: dict file ftp ftps gopher gophers http https imap imaps ldap ldaps mqtt pop3 pop3s rtsp smb smbs smtp smtps telnet tftp
Features: alt-svc AsynchDNS GSS-API HSTS HTTP2 HTTPS-proxy IPv6 Kerberos Largefile libz MultiSSL NTLM NTLM_WB SPNEGO SSL threadsafe UnixSockets

cmake version 3.26.4
CMake suite maintained and supported by Kitware (kitware.com/cmake).
```

`commands`
```
git clone https://github.com/introlab/rtabmap.git rtabmap
cd rtabmap/app/ios/RTABMapApp
./install_deps.sh
```

`error`
```
** BUILD FAILED **

The following build commands failed:
	CompileC /rtabmap-0.21.0/app/ios/RTABMapApp/Libraries/pcl/build/build/pcl_surface.build/Release-iphoneos/Objects-normal/arm64/poisson.o /rtabmap-0.21.0/app/ios/RTABMapApp/Libraries/pcl/surface/src/poisson.cpp normal arm64 c++ com.apple.compilers.llvm.clang.1_0.compiler (in target 'pcl_surface' from project 'PCL')
(1 failure)

matlabbe, 2023-06-11:
Humm, I would have to check later on Mac. Is it possible to have more log about the compilation error?

KeizoJindo, 2023-06-12:
Hello, thank you for developing rtabmap.
I am also trying to create a rtabmap for iOS. I found an error in the same procedure.
I have recently started touching rtabmap and may be making a very elementary mistake. Sorry if I am asking a silly question.

Here are the details:

### Summary

I ran "./install_deps.sh" with reference to the wiki and it seems to be failing to build the Point cloud library.

### environment

* PC: Mac mini M2 pro(2023)
  MacOS 13.4
  cmake 3.26.4
  curl 7.88.1
  git 2.39.2
  xcode 14.3.1
* device: iPad Pro M2 12.9(2022)
* rtabmap v0.21.0

### Terminal

```
+ cmake --build . --config Release
2023-06-12 19:16:04.360 xcodebuild[82499:574111]
DVTCoreDeviceEnabledState: DVTCoreDeviceEnabledState_Disabled set via 
user default (DVTEnableCoreDevice=disabled)
** BUILD FAILED **

The following build commands failed:
CompileC /Users/workdir/rtabmap/app/ios/RTABMapApp/Libraries/pcl/
build/build/pcl_surface.build/Release-iphoneos/Objects-normal/arm64/
poisson.o /Users/workdir/rtabmap/app/ios/RTABMapApp/Libraries/pcl/surface/
src/poisson.cpp normal arm64 c++ 
com.apple.compilers.llvm.clang.1_0.compiler (in target 'pcl_surface' from 
project 'PCL')
(1 failure)
```

### Redirect txt file

The information redirected to a text file from the build results is attached, but only the error portion is excerpted.
[build_log.txt](https://github.com/introlab/rtabmap/files/11722229/build_log.txt)

```
In file included from /Users/workdir/rtabmap/app/ios/RTABMapApp/Libraries/pcl/surface/src/poisson.cpp:40:
In file included from /Users/workdir/rtabmap/app/ios/RTABMapApp/Libraries/pcl/surface/include/pcl/surface/impl/poisson.hpp:49:
In file included from /Users/workdir/rtabmap/app/ios/RTABMapApp/Libraries/pcl/surface/include/pcl/surface/3rdparty/poisson4/sparse_matrix.h:185:
/Users/workdir/rtabmap/app/ios/RTABMapApp/Libraries/pcl/surface/include/pcl/surface/3rdparty/poisson4/sparse_matrix.hpp:256:40: error: member reference base type 'MatrixEntry<T> *' is not a structure or union
        for(int ii=0;ii<m_ppElements[i].size();i++){m_ppElements[i][ii].Value*=V;}
                        ~~~~~~~~~~~~~~~^~~~~
/Users/workdir/rtabmap/app/ios/RTABMapApp/Libraries/pcl/surface/include/pcl/surface/3rdparty/poisson4/sparse_matrix.hpp:266:40: error: member reference base type 'MatrixEntry<T> *' is not a structure or union
        for(int ii=0;ii<m_ppElements[i].size();ii++){
                        ~~~~~~~~~~~~~~~^~~~~
/Users/workdir/rtabmap/app/ios/RTABMapApp/Libraries/pcl/surface/include/pcl/surface/3rdparty/poisson4/sparse_matrix.hpp:327:40: error: member reference base type 'MatrixEntry<T> *' is not a structure or union
        for(int ii=0;ii<m_ppElements[i].size();ii++){
                        ~~~~~~~~~~~~~~~^~~~~
1 warning and 3 errors generated.
```

matlabbe, 2023-06-12:
Thx for the log, we may have to upgrade PCL version in the install script. This error may appear with the more recent clang version.

brunsy, 2023-06-13:
It would make sense to upgrade it as it has been fixed upstream now. 

Although in the meantime @patrickwasp you could always apply it as an extra patch in the install_deps.sh

```
git apply pcl_1_11_1_vtk_ios_support.patch
cp ../../fix-compilation.patch .
git apply fix-compilation.patch
mkdir build
```
fix-compilation.patch (add in the same folder as install_deps.sh)
```
diff --git a/surface/include/pcl/surface/3rdparty/poisson4/sparse_matrix.hpp b/surface/include/pcl/surface/3rdparty/poisson4/sparse_matrix.hpp
index e8fae9698..24f0a5402 100644
--- a/surface/include/pcl/surface/3rdparty/poisson4/sparse_matrix.hpp
+++ b/surface/include/pcl/surface/3rdparty/poisson4/sparse_matrix.hpp
@@ -26,7 +26,6 @@ ANY WAY OUT OF THE USE OF THIS SOFTWARE, EVEN IF ADVISED OF THE POSSIBILITY OF S
 DAMAGE.
 */
 
-#include <float.h>
 #ifdef _WIN32
 # ifndef WIN32_LEAN_AND_MEAN
 #  define WIN32_LEAN_AND_MEAN
@@ -251,9 +250,9 @@ namespace pcl
     template<class T>
     SparseMatrix<T>& SparseMatrix<T>::operator *= (const T& V)
     {
-      for (int i=0; i<this->Rows(); i++)
+      for (int i=0; i<rows; i++)
       {
-        for(int ii=0;ii<m_ppElements[i].size();i++){m_ppElements[i][ii].Value*=V;}
+        for(int ii=0;ii<rowSizes[i];ii++){m_ppElements[i][ii].Value*=V;}
       }
       return *this;
     }
@@ -261,12 +260,12 @@ namespace pcl
     template<class T>
     SparseMatrix<T> SparseMatrix<T>::Multiply( const SparseMatrix<T>& M ) const
     {
-      SparseMatrix<T> R( this->Rows(), M.Columns() );
-      for(int i=0; i<R.Rows(); i++){
-        for(int ii=0;ii<m_ppElements[i].size();ii++){
+      SparseMatrix<T> R( rows, M._maxEntriesPerRow );
+      for(int i=0; i<R.rows; i++){
+        for(int ii=0;ii<rowSizes[i];ii++){
           int N=m_ppElements[i][ii].N;
           T Value=m_ppElements[i][ii].Value;
-          for(int jj=0;jj<M.m_ppElements[N].size();jj++){
+          for(int jj=0;jj<M.rowSizes[N];jj++){
             R(i,M.m_ppElements[N][jj].N) += Value * M.m_ppElements[N][jj].Value;
           }
         }
@@ -320,11 +319,11 @@ namespace pcl
     template<class T>
     SparseMatrix<T> SparseMatrix<T>::Transpose() const
     {
-      SparseMatrix<T> M( this->Columns(), this->Rows() );
+      SparseMatrix<T> M( _maxEntriesPerRow, rows );
 
-      for (int i=0; i<this->Rows(); i++)
+      for (int i=0; i<rows; i++)
       {
-        for(int ii=0;ii<m_ppElements[i].size();ii++){
+        for(int ii=0;ii<rowSizes[i];ii++){
           M(m_ppElements[i][ii].N,i) = m_ppElements[i][ii].Value;
         }
       }

```

KeizoJindo, 2023-06-13:
@brunsy 
Thank you for the fix patch(fix-compilation.patch). The iOS build was successful,  I checked it works on iPad Pro.
(My test environment is as posted above .)
No problem. Thank you!

matlabbe, 2023-06-17:
Based on the patch of @brunsy  and the [commit](https://github.com/PointCloudLibrary/pcl/commit/d253f645f0bfbcd2c818fa5d0a6970c61f39c6b5) I've found on PCL, it would require minimum PCL 1.13.0. I'll make the change.

matlabbe, 2023-06-19:
Increasing to PCL 1.13 was too complicated for the time I have now (the way VTK is included in PCL 1.13 is different than in PCL 1.11, so my original iOS patch doesn't work anymore with 1.13). I integrated the patch into the one used for iOS build and pcl 1.11: https://gist.github.com/matlabbe/f3ba9366eb91e1b855dadd2ddce5746d
The gists version has been updated in in install_deps.sh` in commit above.

commit ab3ade03 (2023-06-19): fixed #1051
closed 2023-06-19 by matlabbe