introlab/rtabmap #1053: Compute covariance during Localization mode

hellovuong, 2023-06-12:
Hello,
I much appreciate your time and patience in resolving/answering the issues, it is an excellent contribution to the community. I open this issue related to the confidence of the localization state. I observe that during the mapping mode, the covariance actually accumulates gets higher over time and reduces by the loop closure. However, in localization mode, it seems like the covariance is always small even in the situation where the robot drifts away from the correct position over time. Can we somehow accumulate the covariance and reduce it with loop closure for localization like in mapping mode? Or in general, can we extract the confidence of the localization state?

hellovuong, 2023-06-12:
Ah I found this answer: https://github.com/introlab/rtabmap_ros/issues/300#issuecomment-1423158724

matlabbe, 2023-06-17:
Note that I recently added that before the first localization, the covariance would be very high > 99. On first localization it will shrink to localization error, then if the robot doesn't localize for a while, the covariance will slowly increase based on odometry covariance. then shrink again on next localization.

closed 2023-06-17 by matlabbe