introlab/rtabmap #1055: rtabmapviz not built

mattiasmar, 2023-06-14:
The rtabmap_ros stderr log says rtabmapviz was not built, since the rtabmap GUI library is missing.
However I built rtabmap using **-DWITH_GUI=ON**.
What could I have gone wrong? What should I check?

_Build command:_
`colcon build --symlink-install --cmake-args -DRTABMAP_SYNC_MULTI_RGBD=ON -DWITH_OPENGV=ON -DCMAKE_BUILD_TYPE=Release  -DCMAKE_CXX_STANDARD=17 -D WITH_TORCH=ON    -DWITH_GUI=ON    -D WITH_PYTHON=ON -D Torch_DIR=/torch/libtorch/share/cmake/Torch -DCMAKE_CXX_FLAGS="-I/usr/include/python3.10/"`

_From the rtabmap_ros stderr.log:_
```
CMake Warning at CMakeLists.txt:526 (MESSAGE):
  Found RTAB-Map built without its GUI library.  Node rtabmapviz will not be
  built!

CMake Warning (dev) at /usr/lib/x86_64-linux-gnu/cmake/Qt5Core/Qt5CoreMacros.cmake:44 (message):
  qt5_use_modules is not part of the official API, and might be removed in Qt
  6.
Call Stack (most recent call first):
  /usr/lib/x86_64-linux-gnu/cmake/Qt5Core/Qt5CoreMacros.cmake:431 (_qt5_warn_deprecated)
  CMakeLists.txt:642 (QT5_USE_MODULES)
This warning is for project developers.  Use -Wno-dev to suppress it.

CMake Warning:
  Manually-specified variables were not used by the project:

    Torch_DIR
    WITH_OPENGV
    WITH_PYTHON
    WITH_TORCH
```

_From the rtabamp stdout.log, all lines with the string 'gui' (looks OK to me, but I might be missing something):_
```
	Line  20: [ 51%] [32mBuilding CXX object guilib/src/CMakeFiles/rtabmap_gui.dir/PreferencesDialog.cpp.o[0m
	Line  21: [ 51%] [32mBuilding CXX object guilib/src/CMakeFiles/rtabmap_gui.dir/MainWindow.cpp.o[0m
	Line  22: [ 51%] [32mBuilding CXX object guilib/src/CMakeFiles/rtabmap_gui.dir/AboutDialog.cpp.o[0m
	Line  23: [ 52%] [32mBuilding CXX object guilib/src/CMakeFiles/rtabmap_gui.dir/DatabaseViewer.cpp.o[0m
	Line  24: [ 52%] [32mBuilding CXX object guilib/src/CMakeFiles/rtabmap_gui.dir/CloudViewer.cpp.o[0m
	Line  25: [ 52%] [32mBuilding CXX object guilib/src/CMakeFiles/rtabmap_gui.dir/OdometryViewer.cpp.o[0m
	Line  26: [ 53%] [32mBuilding CXX object guilib/src/CMakeFiles/rtabmap_gui.dir/LoopClosureViewer.cpp.o[0m
	Line  27: [ 53%] [32mBuilding CXX object guilib/src/CMakeFiles/rtabmap_gui.dir/CameraViewer.cpp.o[0m
	Line  28: [ 53%] [32mBuilding CXX object guilib/src/CMakeFiles/rtabmap_gui.dir/CalibrationDialog.cpp.o[0m
	Line  29: [ 53%] [32mBuilding CXX object guilib/src/CMakeFiles/rtabmap_gui.dir/ExportBundlerDialog.cpp.o[0m
	Line  30: [ 55%] [32mBuilding CXX object guilib/src/CMakeFiles/rtabmap_gui.dir/GraphViewer.cpp.o[0m
	Line  31: [ 56%] [32mBuilding CXX object guilib/src/CMakeFiles/rtabmap_gui.dir/ExportCloudsDialog.cpp.o[0m
	Line  32: [ 56%] [32mBuilding CXX object guilib/src/CMakeFiles/rtabmap_gui.dir/DepthCalibrationDialog.cpp.o[0m
	Line  33: [ 57%] [32mBuilding CXX object guilib/src/CMakeFiles/rtabmap_gui.dir/MultiSessionLocWidget.cpp.o[0m
	Line  34: [ 57%] [32mBuilding CXX object guilib/src/CMakeFiles/rtabmap_gui.dir/__/include/rtabmap/gui/moc_DepthCalibrationDialog.cpp.o[0m
	Line  53: [ 59%] [32m[1mLinking CXX shared library ../../bin/librtabmap_gui.so[0m
	Line  54: [ 90%] Built target rtabmap_gui
	Line 291: -- Installing: /root/ws/install/rtabmap/lib/librtabmap_gui.so.0.21.1
	Line 292: -- Up-to-date: /root/ws/install/rtabmap/lib/librtabmap_gui.so.0.21
	Line 293: -- Set runtime path of "/root/ws/install/rtabmap/lib/librtabmap_gui.so.0.21.1" to ""
	Line 294: -- Up-to-date: /root/ws/install/rtabmap/lib/librtabmap_gui.so
	Line 296: -- Up-to-date: /root/ws/install/rtabmap/include/rtabmap-0.21/rtabmap/gui
	Line 297: -- Up-to-date: /root/ws/install/rtabmap/include/rtabmap-0.21/rtabmap/gui/ProgressDialog.h
	Line 298: -- Up-to-date: /root/ws/install/rtabmap/include/rtabmap-0.21/rtabmap/gui/GraphViewer.h
	Line 299: -- Up-to-date: /root/ws/install/rtabmap/include/rtabmap-0.21/rtabmap/gui/LoopClosureViewer.h
	Line 300: -- Up-to-date: /root/ws/install/rtabmap/include/rtabmap-0.21/rtabmap/gui/MultiSessionLocWidget.h
	Line 301: -- Up-to-date: /root/ws/install/rtabmap/include/rtabmap-0.21/rtabmap/gui/CloudViewerInteractorStyle.h
	Line 302: -- Up-to-date: /root/ws/install/rtabmap/include/rtabmap-0.21/rtabmap/gui/MapVisibilityWidget.h
	Line 303: -- Up-to-date: /root/ws/install/rtabmap/include/rtabmap-0.21/rtabmap/gui/OdometryViewer.h
	Line 304: -- Up-to-date: /root/ws/install/rtabmap/include/rtabmap-0.21/rtabmap/gui/DataRecorder.h
	Line 305: -- Up-to-date: /root/ws/install/rtabmap/include/rtabmap-0.21/rtabmap/gui/CreateSimpleCalibrationDialog.h
	Line 306: -- Up-to-date: /root/ws/install/rtabmap/include/rtabmap-0.21/rtabmap/gui/StatsToolBox.h
	Line 307: -- Up-to-date: /root/ws/install/rtabmap/include/rtabmap-0.21/rtabmap/gui/EditConstraintDialog.h
	Line 308: -- Up-to-date: /root/ws/install/rtabmap/include/rtabmap-0.21/rtabmap/gui/DepthCalibrationDialog.h
	Line 309: -- Up-to-date: /root/ws/install/rtabmap/include/rtabmap-0.21/rtabmap/gui/ParametersToolBox.h
	Line 310: -- Up-to-date: /root/ws/install/rtabmap/include/rtabmap-0.21/rtabmap/gui/PreferencesDialog.h
	Line 311: -- Up-to-date: /root/ws/install/rtabmap/include/rtabmap-0.21/rtabmap/gui/ExportCloudsDialog.h
	Line 312: -- Up-to-date: /root/ws/install/rtabmap/include/rtabmap-0.21/rtabmap/gui/PostProcessingDialog.h
	Line 313: -- Up-to-date: /root/ws/install/rtabmap/include/rtabmap-0.21/rtabmap/gui/CameraViewer.h
	Line 314: -- Up-to-date: /root/ws/install/rtabmap/include/rtabmap-0.21/rtabmap/gui/ConsoleWidget.h
	Line 315: -- Up-to-date: /root/ws/install/rtabmap/include/rtabmap-0.21/rtabmap/gui/EditMapArea.h
	Line 316: -- Up-to-date: /root/ws/install/rtabmap/include/rtabmap-0.21/rtabmap/gui/MainWindow.h
	Line 317: -- Up-to-date: /root/ws/install/rtabmap/include/rtabmap-0.21/rtabmap/gui/PdfPlot.h
	Line 318: -- Up-to-date: /root/ws/install/rtabmap/include/rtabmap-0.21/rtabmap/gui/ExportDialog.h
	Line 319: -- Up-to-date: /root/ws/install/rtabmap/include/rtabmap-0.21/rtabmap/gui/CalibrationDialog.h
	Line 320: -- Up-to-date: /root/ws/install/rtabmap/include/rtabmap-0.21/rtabmap/gui/CloudViewerCellPicker.h
	Line 321: -- Up-to-date: /root/ws/install/rtabmap/include/rtabmap-0.21/rtabmap/gui/MultiSessionLocSubView.h
	Line 322: -- Up-to-date: /root/ws/install/rtabmap/include/rtabmap-0.21/rtabmap/gui/AboutDialog.h
	Line 323: -- Up-to-date: /root/ws/install/rtabmap/include/rtabmap-0.21/rtabmap/gui/RecoveryState.h
	Line 324: -- Up-to-date: /root/ws/install/rtabmap/include/rtabmap-0.21/rtabmap/gui/ImageView.h
	Line 325: -- Up-to-date: /root/ws/install/rtabmap/include/rtabmap-0.21/rtabmap/gui/CloudViewer.h
	Line 326: -- Up-to-date: /root/ws/install/rtabmap/include/rtabmap-0.21/rtabmap/gui/TexturingState.h
	Line 327: -- Up-to-date: /root/ws/install/rtabmap/include/rtabmap-0.21/rtabmap/gui/KeypointItem.h
	Line 328: -- Up-to-date: /root/ws/install/rtabmap/include/rtabmap-0.21/rtabmap/gui/EditDepthArea.h
	Line 329: -- Up-to-date: /root/ws/install/rtabmap/include/rtabmap-0.21/rtabmap/gui/DatabaseViewer.h
	Line 330: -- Up-to-date: /root/ws/install/rtabmap/include/rtabmap-0.21/rtabmap/gui/ExportBundlerDialog.h
	Line 336: -- Up-to-date: /root/ws/install/rtabmap/include/rtabmap-0.21/rtabmap/gui
	Line 337: -- Installing: /root/ws/install/rtabmap/include/rtabmap-0.21/rtabmap/gui/rtabmap_gui_export.h
	Line 338: -- Installing: /root/ws/install/rtabmap/lib/rtabmap-0.21/RTABMap_guiTargets.cmake
	Line 339: -- Installing: /root/ws/install/rtabmap/lib/rtabmap-0.21/RTABMap_guiTargets-release.cmake
```

matlabbe, 2023-06-17:
Which `rtabmap_ros` version are you using?

mattiasmar, 2023-06-18:
rtabmap_ros: 0.20.23

mattiasmar, 2023-06-18:
I updated to the head of rtabmap_ros and now rtabmapviz runs. Thanks!

closed 2023-06-18 by mattiasmar