introlab/rtabmap #1064: Superpoint seg fault NMS

matlabbe, 2023-06-20:
```
Thread 53 "rtabmap" received signal SIGSEGV, Segmentation fault.
[Switching to Thread 0x7ffefdffb000 (LWP 132998)]
0x00007ffff73eeb53 in rtabmap::NMS(std::vector<cv::KeyPoint, std::allocator<cv::KeyPoint> > const&, cv::Mat const&, cv::Mat const&, std::vector<cv::KeyPoint, std::allocator<cv::KeyPoint> >&, cv::Mat&, int, int, int, int) () from /home/mathieu/workspace/rtabmap/build/bin/librtabmap_core.so.0.21
(gdb) bt
#0  0x00007ffff73eeb53 in rtabmap::NMS(std::vector<cv::KeyPoint, std::allocator<cv::KeyPoint> > const&, cv::Mat const&, cv::Mat const&, std::vector<cv::KeyPoint, std::allocator<cv::KeyPoint> >&, cv::Mat&, int, int, int, int) () at /home/mathieu/workspace/rtabmap/build/bin/librtabmap_core.so.0.21
#1  0x00007ffff73efc9d in rtabmap::SPDetector::detect(cv::Mat const&, cv::Mat const&) () at /home/mathieu/workspace/rtabmap/build/bin/librtabmap_core.so.0.21
#2  0x00007ffff7049545 in rtabmap::SuperPointTorch::generateKeypointsImpl(cv::Mat const&, cv::Rect_<int> const&, cv::Mat const&) ()
    at /home/mathieu/workspace/rtabmap/build/bin/librtabmap_core.so.0.21
#3  0x00007ffff7053b72 in rtabmap::Feature2D::generateKeypoints(cv::Mat const&, cv::Mat const&) () at /home/mathieu/workspace/rtabmap/build/bin/librtabmap_core.so.0.21
#4  0x00007ffff7253133 in rtabmap::RegistrationVis::computeTransformationImpl(rtabmap::Signature&, rtabmap::Signature&, rtabmap::Transform, rtabmap::RegistrationInfo&) const ()
    at /home/mathieu/workspace/rtabmap/build/bin/librtabmap_core.so.0.21
#5  0x00007ffff7227ec0 in rtabmap::Registration::computeTransformationMod(rtabmap::Signature&, rtabmap::Signature&, rtabmap::Transform, rtabmap::RegistrationInfo*) const ()
    at /home/mathieu/workspace/rtabmap/build/bin/librtabmap_core.so.0.21
#6  0x00007ffff72a635b in rtabmap::OdometryF2M::computeTransform(rtabmap::SensorData&, rtabmap::Transform const&, rtabmap::OdometryInfo*) ()
    at /home/mathieu/workspace/rtabmap/build/bin/librtabmap_core.so.0.21
#7  0x00007ffff7296834 in rtabmap::Odometry::process(rtabmap::SensorData&, rtabmap::Transform const&, rtabmap::OdometryInfo*) () at /home/mathieu/workspace/rtabmap/build/bin/librtabmap_core.so.0.21
#8  0x00007ffff729a6eb in rtabmap::Odometry::process(rtabmap::SensorData&, rtabmap::OdometryInfo*) () at /home/mathieu/workspace/rtabmap/build/bin/librtabmap_core.so.0.21
#9  0x00007ffff72a1df2 in rtabmap::OdometryThread::mainLoop() () at /home/mathieu/workspace/rtabmap/build/bin/librtabmap_core.so.0.21
#10 0x00007ffff6abc891 in UThread::ThreadMain() () at /home/mathieu/workspace/rtabmap/build/bin/librtabmap_utilite.so.0.21
#11 0x00007ffff6abcbb9 in UThreadC<void>::ThreadMainHandler(UThreadC<void>::Instance*) () at /home/mathieu/workspace/rtabmap/build/bin/librtabmap_utilite.so.0.21
#12 0x00007ffff1c1a609 in start_thread (arg=<optimized out>) at pthread_create.c:477
#13 0x00007ffff5ba2133 in clone () at ../sysdeps/unix/sysv/linux/x86_64/clone.S:95
```
We can see the first frame with keypoints, but crash the next one. On one test, I could see 15-20 frames until it crashes. Not sure if it is related to keypoints extracted from the black borders (tested with OAK-D).

matlabbe, 2023-06-20:
Same error with D435:
```
Thread 60 "rtabmap" received signal SIGSEGV, Segmentation fault.
[Switching to Thread 0x7ffee17fe000 (LWP 133909)]
0x00007ffff73eeb53 in rtabmap::NMS(std::vector<cv::KeyPoint, std::allocator<cv::KeyPoint> > const&, cv::Mat const&, cv::Mat const&, std::vector<cv::KeyPoint, std::allocator<cv::KeyPoint> >&, cv::Mat&, int, int, int, int) () from /home/mathieu/workspace/rtabmap/build/bin/librtabmap_core.so.0.21
(gdb) bt
#0  0x00007ffff73eeb53 in rtabmap::NMS(std::vector<cv::KeyPoint, std::allocator<cv::KeyPoint> > const&, cv::Mat const&, cv::Mat const&, std::vector<cv::KeyPoint, std::allocator<cv::KeyPoint> >&, cv::Mat&, int, int, int, int) () at /home/mathieu/workspace/rtabmap/build/bin/librtabmap_core.so.0.21
#1  0x00007ffff73efc9d in rtabmap::SPDetector::detect(cv::Mat const&, cv::Mat const&) () at /home/mathieu/workspace/rtabmap/build/bin/librtabmap_core.so.0.21
#2  0x00007ffff7049545 in rtabmap::SuperPointTorch::generateKeypointsImpl(cv::Mat const&, cv::Rect_<int> const&, cv::Mat const&) ()
    at /home/mathieu/workspace/rtabmap/build/bin/librtabmap_core.so.0.21
#3  0x00007ffff7053b72 in rtabmap::Feature2D::generateKeypoints(cv::Mat const&, cv::Mat const&) () at /home/mathieu/workspace/rtabmap/build/bin/librtabmap_core.so.0.21
#4  0x00007ffff724fbd0 in rtabmap::RegistrationVis::computeTransformationImpl(rtabmap::Signature&, rtabmap::Signature&, rtabmap::Transform, rtabmap::RegistrationInfo&) const ()
    at /home/mathieu/workspace/rtabmap/build/bin/librtabmap_core.so.0.21
#5  0x00007ffff7227ec0 in rtabmap::Registration::computeTransformationMod(rtabmap::Signature&, rtabmap::Signature&, rtabmap::Transform, rtabmap::RegistrationInfo*) const ()
    at /home/mathieu/workspace/rtabmap/build/bin/librtabmap_core.so.0.21
#6  0x00007ffff72a48c3 in rtabmap::OdometryF2M::computeTransform(rtabmap::SensorData&, rtabmap::Transform const&, rtabmap::OdometryInfo*) ()
    at /home/mathieu/workspace/rtabmap/build/bin/librtabmap_core.so.0.21
#7  0x00007ffff7296834 in rtabmap::Odometry::process(rtabmap::SensorData&, rtabmap::Transform const&, rtabmap::OdometryInfo*) () at /home/mathieu/workspace/rtabmap/build/bin/librtabmap_core.so.0.21
#8  0x00007ffff729a6eb in rtabmap::Odometry::process(rtabmap::SensorData&, rtabmap::OdometryInfo*) () at /home/mathieu/workspace/rtabmap/build/bin/librtabmap_core.so.0.21
#9  0x00007ffff72a1df2 in rtabmap::OdometryThread::mainLoop() () at /home/mathieu/workspace/rtabmap/build/bin/librtabmap_core.so.0.21
#10 0x00007ffff6abc891 in UThread::ThreadMain() () at /home/mathieu/workspace/rtabmap/build/bin/librtabmap_utilite.so.0.21
#11 0x00007ffff6abcbb9 in UThreadC<void>::ThreadMainHandler(UThreadC<void>::Instance*) () at /home/mathieu/workspace/rtabmap/build/bin/librtabmap_utilite.so.0.21
#12 0x00007ffff1c1a609 in start_thread (arg=<optimized out>) at pthread_create.c:477
#13 0x00007ffff5ba2133 in clone () at ../sysdeps/unix/sysv/linux/x86_64/clone.S:95
```
There could be a bug in NMS function.

matlabbe, 2023-06-21:
It was caused by increasing nms distance threshold parameters at 16 (default 4). This issue will be fixed after #1058 is merged

matlabbe, 2023-06-21:
Fixed by commit above. Closing.

closed 2023-06-21 by matlabbe

mattiasmar, 2023-06-21:
I will pull and try it.  Is the fastest CPU performance achieved with torch or with depthai's  blobs (if at all relevant for CPU)?
What latencies have you measured for superpoint and superglue with CPU only?

borongyuan, 2023-06-21:
Hi @mattiasmar 
The SuperPoint blob is a hardware-specific feature that runs on OAK cameras. I haven't compared its performance with Torch CPU yet. But it’s actually based on OpenVINO. So it should be possible to change it to OpenVINO implementation to run using different backends. If more neural network related functions are to be added to RTAB-Map later, it may be more versatile to use a framework such as ONNXRuntime or TVM. This makes it easy to use different backends with better performance. The general situation is that Myriad VPU is slower than CPU. But if you use Intel GPU for inference, it will be much better than using CPU.

mattiasmar, 2023-06-21:
@borongyuan 
Is there a quick way for me to test its performance on CPU?

matlabbe, 2023-06-22:
@mattiasmar Cannot test superglue now on CPU, but for superpoint, here is the average odometry time (on host) with OAK 400p resolution:
 * depthai-superpoint (superpoint computed on OAK-D): 30 ms
 * superpoint torch CPU (11th Gen Intel® Core™ i7-11700KF @ 3.60GHz × 16): 180 ms
 * superpoint torch GPU (NVIDIA GeForce RTX 3080): 60 ms

mattiasmar, 2023-06-22:
Thanks. But one could execute the depthai-superpoint too on CPU, couldn't one?

borongyuan, 2023-06-22:
@mattiasmar 
https://github.com/jinac/superpoint_infer_engine