introlab/rtabmap #1065: [Feature-Request] NVIDIA VPI Support

borongyuan, 2023-06-21:
Hi,
I'm interested in adding support for NVIDIA  [Vision Programming Interface (VPI)](https://developer.nvidia.com/embedded/vpi). It provides many [algorithms](https://docs.nvidia.com/vpi/algorithms.html) similar to OpenCV, but can be accelerated using hardware processing units on the Jetson platform. It also supports x86 platform.
It seems that there are not many people using VPI now, and I don't know how far it can go. NVIDIA released VisionWorks a few years ago, and then dropped it. But now VPI has been updated to 2.2, many functions seem to be very useful.

matlabbe, 2023-06-22:
Some features of VPI that could replace opencv functions used in rtabmap:

- [Stereo Disparity Estimator](https://docs.nvidia.com/vpi/algo_stereo_disparity.html)
- [KLT Feature Tracker](https://docs.nvidia.com/vpi/algo_klt_tracker.html)
- [Harris Corner Detector](https://docs.nvidia.com/vpi/algo_harris_corners.html)
- [Pyramidal LK Optical Flow](https://docs.nvidia.com/vpi/algo_optflow_lk.html)
- [FAST Corners Detector](https://docs.nvidia.com/vpi/algo_fast_corners_detector.html)
- [ORB feature detector](https://docs.nvidia.com/vpi/algo_orb_feature_detector.html)

Looking at [C++ API](https://docs.nvidia.com/vpi/tutorial_c.html), it looks not too difficult to integrate.

About NVidia, another library I wanted to try eventually (in particular for fast VO on jetson): https://docs.nvidia.com/isaac/packages/visual_slam/doc/elbrus_visual_slam.html

borongyuan, 2023-06-22:
> About NVidia, another library I wanted to try eventually (in particular for fast VO on jetson): https://docs.nvidia.com/isaac/packages/visual_slam/doc/elbrus_visual_slam.html

You can try their custom SD card image for Jetson Orin Nano. I just reflashed my Jetson a few days ago.
https://nvidia-ai-iot.github.io/jetson_isaac_ros_visual_slam_tutorial/sdcard.html
This is convenient for quick testing without the hassle of configuring the environment.