introlab/rtabmap #1067: CMake Error IMPORTED_LOCATION not set for imported target "octomap" configuration "Release".

Userpc1010, 2023-06-23:
[image]

Temporary solution in CMakeLists.txt:

```
#    FIND_PACKAGE(octomap QUIET)
    set (octomap_FOUND true)
    set (octomap_VERSION 1.9.8)
    set (OCTOMAP_INCLUDE_DIRS /usr/local/include)
    set (OCTOMAP_LIBRARY_DIRS /usr/local/lib)
    set (OCTOMAP_LIBRARIES /usr/local/lib/liboctomap.so;/usr/local/lib/liboctomath.so)
```
[image]

matlabbe, 2023-06-25:
Not sure why you had the error, unless it is an octomap bug. For recent octomap, we use by default cmake targets if they exist instead of LIBRARIES/INCLUDE_DIRS:
https://github.com/introlab/rtabmap/blob/092f6abccc0f19ad0bc8397ccd723e3de34ebd28/corelib/src/CMakeLists.txt#L570-L590

In your case, it means it was able to find octomap target, but the target has missing IMPORTED_LOCATION. I see you built octomap from source, which commit version are you using? (here with ros noetic 1.9.8 octomap version, it works correctly)

Userpc1010, 2023-06-26:
I used the latest version currently available [from the repository](https://github.com/OctoMap/octomap.git). Before this, I had a similar error when building [No rule to make target 'octomap-NOTFOUND'](https://github.com/flexible-collision-library/fcl/issues/210#issuecomment-320105117)

matlabbe, 2023-06-29:
What is your system? Just tried on ubuntu 20.04, cloning that repo, install it, remove ros-noetic-octomap, cleanup rtabmap's build directory, then do cmake. I didn't get error in cmake log, but one when the compiler tries linking `librtabmap_core.so`:
```
make[2]: *** No rule to make target 'octomap-NOTFOUND', needed by 'bin/librtabmap_core.so.0.21.1'.  Stop.
make[2]: *** Waiting for unfinished jobs....
make[1]: *** [CMakeFiles/Makefile2:936: corelib/src/CMakeFiles/rtabmap_core.dir/all] Error 2
make: *** [Makefile:152: all] Error 2
```

The problem is that octomap doesn't set a default for `CMAKE_BUILD_TYPE`, so not sure if it is why the target is not correctly exported. Rebuilt/installed octomap in release mode fixed the issue:
```
cd octomap/build
cmake -DCMAKE_BUILD_TYPE=Release ..
sudo make install
```

Userpc1010, 2023-06-29:
I used Ubuntu 20.04, probably this was the problem, I just repeated the input from the README without  `-DCMAKE_BUILD_TYPE=Release`.

closed 2023-06-29 by Userpc1010