introlab/rtabmap #1071: Is it ok to only use RGB image and depth image? No imu, no lidar, no stereo

TwiceMao, 2023-06-27:
I now have the rgb image and the corresponding depth image, as well as the internal parameters of the camera. I want to use rtabmap as an odometry to estimate the camera pose of each frame. I don't know if it is feasible? Oh, yes, it is in the ros2 environment.

If so，could you give me some examples，or advice？ Thx

matlabbe, 2023-06-27:
Assuming you have topics `/camera/rgb/image_rect_color` as rgb image, `/camera/depth_registered/image_raw` as registered depth image to rgb camera and `/camera/rgb/camera_info` as the rgb camera calibration, you can do (adjust `qos` if needed):

```
ros2 launch rtabmap_launch rtabmap.launch.py \
    rtabmap_args:="--delete_db_on_start" \
    rgb_topic:=/camera/rgb/image_rect_color \
    depth_topic:=/camera/depth_registered/image_raw \
    camera_info_topic:=/camera/rgb/camera_info \
    frame_id:=base_link \
    approx_sync:=false \
    qos:=1 \
    rviz:=true
```

TwiceMao, 2023-07-03:
What parameters are used to modify the scale of the depth map? The scale of our depth map is equal to 63335.

matlabbe, 2023-07-04:
if depth is stored as 16UC1, rtabmap assumes it is in mm. There is no scaling parameter. For 32FC1 format, it assumes it is in meters. If you have a scale problem, fix camera calibration.

To view estimated pose, check TF in rviz. The odometry pose is published on `/rtabmap/odom` topic.

TwiceMao, 2023-07-04:
1. When the depth map in the ros2 bag uses '32FC1' (the unit is m, which matches the camera parameters ), the visualization result of using rtabmap as odometry is shown in the figure below. Strangely, the Loop closure detection section is  empty.
[image]
[image]

2. When the depth map in the ros2 bag uses '16UC1' (it does not match the camera parameters at this time, you need to divide the pixel value by 6553.5 to match). It is very strange that not only the depth error cannot be seen from the 3D Map, but also the Loop closure detection can find a matching that looks correct. But at this time, there is only the camera trajectory map (with loop matching) in RViz, but there is no 3D map.
[image]
[image]
Why？
3. When I set the qos parameter to 0, rtabmap cannot start. If the qos parameter is set to 1, it can be started normally. Why?

matlabbe, 2023-07-07:
1. the depth format should not have any effect on loop closure detection. 

2.
> (it does not match the camera parameters at this time, you need to divide the pixel value by 6553.5 to match)

not sure to understand this, camera parameters should be independent of the depth units and they should always be in meters. When receiving 16UC1, rtabmap will divide the depth pixel value by 1000 to convert to meters.

For rviz, not sure why the 3D map worked the first example and not this one.

3. look at https://docs.ros.org/en/rolling/Concepts/Intermediate/About-Quality-of-Service-Settings.html. For this parameter in rtabmap, here are the possible values: 0=system default, 1=Reliable, 2=Best Effort.

4. it depends in which frame you are estimating the pose. If your TF is standard (`/base_link->/camera_link->/camera_optical_frame`), and rtabmap use `frame_id` as `base_link`, the pose is following [REP105](https://www.ros.org/reps/rep-0105.html)

closed 2023-07-07 by TwiceMao