introlab/rtabmap #1083: Map Loading

takahiko-hasegawa-hdjp, 2023-07-04:
Hi, 

I am currently trying to develop an QT application that can load map.db and display the map on the GUI.

However, the map data stored in the database is the "unoptimized" version of the map; I need to optimize the map everyttime the map gets loaded. 

I suppose I need to use the existing rtabmap library to optimize the map but not to sure where to look at. 

Could you give me some advice on how to do that?

thanks in advance

matlabbe, 2023-07-07:
You can look for `opt_poses` in Admin table to get already optimized graph, for convenience using `DBDriver` class:
https://github.com/introlab/rtabmap/blob/64f79813cd2c3a7e3f916a7ef2637e93f1f13a5f/corelib/include/rtabmap/core/DBDriver.h#L108

takahiko-hasegawa-hdjp, 2023-08-01:
thanks! 
how about optimized 2D point cloud?

matlabbe, 2023-08-05:
Do you mean the oocupancy grid? it is also in Admin table (see [here](https://github.com/introlab/rtabmap/blob/e2f037189a6963296dc3752ee51437d0ea5676dc/corelib/src/resources/DatabaseSchema.sql.in#L130C2-L133)), though only created automatically on ROS when there is a subscriber on the map. Otherwise, you can regenerate the occupancy grid in `rtabmap-databaseViewer` File->Renegerate optimized 2d map...