introlab/rtabmap #1088: Motion estimation: current implementation of Non-central absolute pose estimation has high chance of using correspondence from only 1 camera

guoqingzh, 2023-07-07:
We have 3 RGB-D cameras sending data to Rtabmap, after matching all 3 cameras has some feature points but the Ransac inlier is very likely (not always but with high chance) all from only 1 camera. My current solution is to increase the Vis/MinInlier to reduce the chance. 

My questions:
 - It seems the current GP3P + Ransac implementation is very likely to bias 1 camera for correspondence so we are not benefiting too much from the multicam setup. Shall we use GPnP with all correspondence ? or make the number of correspondence configurable ?

matlabbe, 2023-07-10:
So you are referring to this code:
https://github.com/introlab/rtabmap/blob/64f79813cd2c3a7e3f916a7ef2637e93f1f13a5f/corelib/src/util3d_motion_estimation.cpp#L345-L370

From OpenGV doc, not sure we can use other than GP3P for non-central problem: 
```
TWOPT = 0,     // central, with rotation prior
KNEIP = 1,     // central     [1]
GAO = 2,       // central     [2]
EPNP = 3,      // central     [4]
GP3P = 4       // non-central [3]
```

If GP3P is indeed sampling randomly across all cameras, instead of uniformly sampling across all cameras, it may have a bias on one camera. From OpenGV [documentation](https://laurentkneip.github.io/opengv/page_how_to_use.html):

> The MultiNoncentralRelativePoseSacProblem is for non-central, non-overlapping viewpoints with little change, and exploits the grouping in order to do homogeneous sampling of correspondences over the cameras. As an example, imagine we are computing the relative pose of a non-overlapping multi-camera rig with two cameras facing opposite directions. In terms of accuracy, it doesn't make sense to sample 16-points in one camera and one point in the other. We preferrably would like to sample 8 points in one camera, and 9 in the other. This is exactly what MultiNoncentralRelativePoseSacProblem is able to do. It uses the derived adapter NoncentralRelativeMultiAdapter.

We implementated `AbsolutePoseSacProblem`, though this note is about `MultiNoncentralRelativePoseSacProblem`. I think we may need to use [MultiNoncentralAbsolutePoseSacProblem](
https://laurentkneip.github.io/opengv/MultiNoncentralAbsolutePoseSacProblem_8hpp.html):

> Functions for fitting an absolute-pose model to a set of bearing-vector-point correspondences, using different algorithms (central and non-central one). Used in a sample-consenus paradigm for rejecting outlier correspondences, which does homogeneous sampling.

guoqingzh, 2023-07-11:
@matlabbe  PR https://github.com/introlab/rtabmap/pull/1091   Use MultiNoncetralAbsolutePoseSacProblem

matlabbe, 2023-07-11:
Great! I'll take a look.