introlab/rtabmap #1090: DB Viewer crashes with SuperPoint

mattiasmar, 2023-07-09:
Using SuperPoint causes  rtabmap-databaseviewer to crash (segmentation fault) when adding loop closures.

When I try to induce a loop closure in the DB viewer I get this error whenever the SP/SG is called more than once:

Superglue execution times:  0.8277442455291748 [-0.827739953994751]
[ INFO] (2023-07-06 21:01:10.415) PythonInterface.cpp:48::~PythonInterface() Py_Finalize() with thread = 673533952
[ INFO] (2023-07-06 21:01:10.654) DatabaseViewer.cpp:8262::refineConstraint() (1 ->2) Registration time: 1.713461 s
[ INFO] (2023-07-06 21:01:10.680) PythonInterface.cpp:25::PythonInterface() Py_Initialize() with thread = 673533952
[ INFO] (2023-07-06 21:01:10.706) PyMatcher.cpp:33::PyMatcher() path = /root/ws/src/rtabmap/archive/2022-IlluminationInvariant/scripts/SuperGluePretrainedNetwork/rtabmap_superglue.py
[ INFO] (2023-07-06 21:01:10.706) PyMatcher.cpp:34::PyMatcher() model = indoor
Segmentation fault (core dumped)

related too to #896

mattiasmar, 2023-07-09:
More from the logs:
```
[DEBUG] (2023-07-09 13:45:53.064) RegistrationVis.cpp:2125::computeTransformationImpl() transform=xyz=-0.001680,-0.001711,0.000000 rpy=0.000000,-0.000000,0.000357
[DEBUG] (2023-07-09 13:45:53.065) PythonInterface.cpp:58::lock() Lock: Current thread=351391744 (main=351391744)
[DEBUG] (2023-07-09 13:45:53.065) PythonInterface.cpp:83::unlock() Unlock: Current thread=351391744 (main=351391744)
[ INFO] (2023-07-09 13:45:53.065) PythonInterface.cpp:48::~PythonInterface() Py_Finalize() with thread = 351391744
[DEBUG] (2023-07-09 13:45:53.295) DatabaseViewer.cpp:8606::addConstraint() 
[DEBUG] (2023-07-09 13:45:53.296) DatabaseViewer.cpp:8938::updateLinksWithModifications() linksAdded_=0 linksRefined_=0 linksRemoved_=0
[DEBUG] (2023-07-09 13:45:53.297) Optimizer.cpp:196::getConnectedGraph() IN: fromId=1 poses=228 links=541 priorsIgnored=1 landmarksIgnored=0
[DEBUG] (2023-07-09 13:45:53.307) Optimizer.cpp:301::getConnectedGraph() OUT: poses=210 links=271
[DEBUG] (2023-07-09 13:45:53.308) OptimizerGTSAM.cpp:101::optimize() Optimizing graph...
[DEBUG] (2023-07-09 13:45:53.308) OptimizerGTSAM.cpp:146::optimize() hasGPSPrior=false
[DEBUG] (2023-07-09 13:45:53.308) OptimizerGTSAM.cpp:163::optimize() fill poses to gtsam... rootId=1 (priorsIgnored=1 landmarksIgnored=0)
[DEBUG] (2023-07-09 13:45:53.308) OptimizerGTSAM.cpp:223::optimize() fill edges to gtsam...
[DEBUG] (2023-07-09 13:45:53.312) OptimizerGTSAM.cpp:495::optimize() create optimizer
[DEBUG] (2023-07-09 13:45:53.314) OptimizerGTSAM.cpp:520::optimize() GTSAM optimizing begin (max iterations=20, robust=0)
[DEBUG] (2023-07-09 13:45:53.322) OptimizerGTSAM.cpp:604::optimize() iteration 1 error =4.261165
[DEBUG] (2023-07-09 13:45:53.329) OptimizerGTSAM.cpp:604::optimize() iteration 2 error =4.261164
[DEBUG] (2023-07-09 13:45:53.329) OptimizerGTSAM.cpp:614::optimize() Stop optimizing, not enough improvement (0.000001 < 0.000010)
[DEBUG] (2023-07-09 13:45:53.332) OptimizerGTSAM.cpp:633::optimize() GTSAM optimizing end (2 iterations done, error=4.261164 (initial=4.262749 final=4.261164), time=0.015075 s)
[DEBUG] (2023-07-09 13:45:53.335) OptimizerGTSAM.cpp:697::optimize() Computing marginals...
[DEBUG] (2023-07-09 13:45:53.341) OptimizerGTSAM.cpp:701::optimize() Computed marginals = 0.006103s (key=228)
[DEBUG] (2023-07-09 13:45:53.342) OptimizerGTSAM.cpp:750::optimize() Optimizing graph...end!
[DEBUG] (2023-07-09 13:45:53.342) Graph.cpp:912::computeMaxGraphErrors() poses=210 links=271
[ INFO] (2023-07-09 13:45:53.354) DatabaseViewer.cpp:8701::addConstraint() Max optimization linear error = 1.122257 m (link 177->167, var=0.449315, ratio error/std=1.674237)
[ INFO] (2023-07-09 13:45:53.354) DatabaseViewer.cpp:8720::addConstraint() Max optimization angular error = 3.266062 deg (link 177->176, var=0.005420, ratio error/std=0.774298)
[ INFO] (2023-07-09 13:45:53.355) DatabaseViewer.cpp:4217::detectMoreLoopClosures() Added new loop closure between 32 and 1.
[DEBUG] (2023-07-09 13:45:53.355) ProgressDialog.cpp:104::appendText() Detected loop closure 32->1! (1/4048)
[DEBUG] (2023-07-09 13:45:53.362) Registration.cpp:49::create() type=0
[ INFO] (2023-07-09 13:45:53.363) PythonInterface.cpp:25::PythonInterface() Py_Initialize() with thread = 351391744
[ INFO] (2023-07-09 13:45:53.385) PyMatcher.cpp:33::PyMatcher() path = /root/ws/src/rtabmap/archive/2022-IlluminationInvariant/scripts/SuperGluePretrainedNetwork/rtabmap_superglue.py
[ INFO] (2023-07-09 13:45:53.385) PyMatcher.cpp:34::PyMatcher() model = indoor
[DEBUG] (2023-07-09 13:45:53.385) PythonInterface.cpp:58::lock() Lock: Current thread=351391744 (main=351391744)
[DEBUG] (2023-07-09 13:45:53.400) PyMatcher.cpp:55::PyMatcher() PyImport_Import
Segmentation fault (core dumped)

```

matlabbe, 2023-07-10:
Yeah, this is a known issue with SuperGlue. It seems there is a bug in rtabmap when accessing the python interpreter from different threads.

mattiasmar, 2023-07-11:
Any idea on how to bridge this problem?

closed 2023-07-13 by mattiasmar