introlab/rtabmap #1096: Question about different poses to export

TakShimoda, 2023-07-24:
Hello, apologies if this question can be easily answered, but I looked through the wiki and tutorials, as well as issues in this repository but found no definitive answers.

I'm running on ROS2 foxy on ubuntu 20.04 and used RTABMAP for RGB-D slam, playing my data from a .db3 bag file. I specifically used the [launch file from here](https://github.com/introlab/rtabmap_ros/blob/foxy-devel/rtabmap_launch/launch/rtabmap.launch.py), using most of the default arguments except changing the topics and ground truth frames to match my bag file.

Here is a sample output, with the grey being ground truth poses provided by a vicon tracking system (in the bag file), and blue being the poses optimized by rtabmap.
[image]

Afterwards, I go to the database with the command `rtabmap-databaseViewer ~/.ros/rtabmap.db`, and then to get the poses, as well as ground truth poses in TUM format I go to File>Export poses>RGBD-SLAM format(*.txt) from which I get a prompt "Which poses?" and it gives me 3 options: 

1. Map's graph (see Graph view) from which I get poses.txt
2. Odometry from which I get poses_odom.txt
3. Ground Truth from which I get poses_gt.txt

For all 3, I chose the Robot (base frame) as the frame as I'm considering the robot's base link as the first child link to my odom frame in the TF tree. 

My main questions are:

1. What is exactly different from options 1 and 2 from above? I assume odometry (poses_odom.txt) would be pure odometry values with no loop closure/long-term correction by the map, but when I plot the trajectories ([using this library to correct alignment](https://github.com/gereon-t/trajectopy)), it is almost the same as the output from option 1(poses.txt) as shown below:
[image]

[image]
I assume they may be similar because the robot doesn't move too great a distance, so even with only odometry, drift doesn't occur.
One thing I do notice is poses.txt only has 55 lines, while poses_odom.txt has 264 lines. You can see from the RPY values above that the "poses" values stop after a certain timeframe.

2. When I look at the poses_gt.txt, I notice it only has 198 lines, although from my bag file, it should have 689 lines. I exported from the bag file to a txt and compared both and they look the same, except poses_gt.txt has fewer lines. Is there a reason for this discrepancy? I tried changing the QOS policy to "best effort" but it really didn't change anything. When comparing the outputs given by rtabmap, should I simply stick with the raw ground truth provided by my bag file or the one given from rtabmap? 
3. As can be seen from the original output from rtabmap, although the poses look accurate, they are offset from the ground truth. Is there a way to fix it with some initialization, or is it due to faulty ground truth data? I know my system may not be calibrated to account for an offset value between the vicon tracking on the robot and the camera's optical center, which I would have to fix on my own end.

Thank you.

matlabbe, 2023-07-31:
1. The poses and poses_odom would be very similar because there is no loop closure in that example. Not sure why you get more odom/gt poses than map's poses, unless `Rtabmap/CreateIntermediateNodes` is used somewhere I don't see (in which case, that is expected). For the "missing" poses at the end, it is because the camera was not moving, thus no new nodes were added to graph.
2. It won't record more gt poses than odometry poses. If you want to keep all gt values, you would have to use your custom solution.
3. As you export the robot poses, you should make sure your vicon is tracking the same base frame, or you will have to offset somehow.

TakShimoda, 2023-07-31:
Thank you, that answers my questions. I figure for 1) there are less poses as it optimizes over every few odometry measurements, which I can see when it runs in real time.

closed 2023-07-31 by TakShimoda