introlab/rtabmap #1098: vcpkg compiles RTabMap project, but without RealSense2 support

MarcoPicciau, 2023-07-24:
I'm trying to compile RTabMap on Windows 10, using vcpkg. 

It compiles RTabMap project, but without RealSense2 support.

The comand sequense is this:

    git clone https://github.com/Microsoft/vcpkg.git
    
    cd vcpkg
    
    bootstrap-vcpkg.bat
    
    vcpkg install realsense2[core] --triplet x64-windows --clean-after-build
    
    vcpkg integrate install
    
    vcpkg install rtabmap --triplet x64-windows --clean-after-build
    
Is necessary particular options o other commands. 

Thank you!

matlabbe, 2023-07-24:
You have to build rtabmap with `realsense2` option:
```
vcpkg install rtabmap[realsense2] --triplet x64-windows --clean-after-build
```

matlabbe, 2023-07-31:
`tm2` option has been [removed](https://github.com/microsoft/vcpkg/commit/62c4a7ad6333ae50b31302c48cb3ca5712828b03) from realsense2 dependency, update rtabmap's vcpkg.conf with those changes:
```patch
diff --git a/ports/rtabmap/vcpkg.json b/ports/rtabmap/vcpkg.json
index ecbc94570..fea10af24 100644
--- a/ports/rtabmap/vcpkg.json
+++ b/ports/rtabmap/vcpkg.json
@@ -117,13 +117,7 @@
     "realsense2": {
       "description": "Build RTAB-Map with RealSense2 camera driver",
       "dependencies": [
-        {
-          "name": "realsense2",
-          "default-features": false,
-          "features": [
-            "tm2"
-          ]
-        }
+        "realsense2"
       ]
     },
     "tools": {
```

MarcoPicciau, 2023-08-29:
thanks, i try it