introlab/rtabmap #1099: 2D GridMap

Hagonata, 2023-07-25:
Someone know how you can achieve some result like this guy with graph view? [https://www.youtube.com/watch?v=NDkw-YhQfXI&t=121s](youtube_url) I have almost the same camera D455, its something with laptop/pc hardware? Or maybe correct launch file?
Help please if you know.

matlabbe, 2023-07-30:
Can you show example of your result to compare? It looks like he used the default parameters.

Hagonata, 2023-08-11:
> Can you show example of your result to compare? It looks like he used the default parameters.

My result cannot overwitre grid like this.

matlabbe, 2023-08-13:
For overwriting cells, make sure `Grid/RayTracing` is on.

matlabbe, 2023-08-18:
Under rtabmap node:
```xml
<param name="Grid/RayTracing" type="string" value="true"/>
```

Hagonata, 2023-08-21:
> `    <include file="$(find rtabmap_ros)/launch/rtabmap.launch">
        <param name="queue_size" value="20"/>
        <param name="Grid/RayTracing" type="string" value="true"/>
        <arg name="args" value="--delete_db_on_start"/>
        <arg name="rgb_topic" value="/camera/color/image_raw"/>
        <arg name="depth_topic" value="/camera/aligned_depth_to_color/image_raw"/>
        <arg name="camera_info_topic" value="/camera/color/camera_info"/>
        <arg name="depth_camera_info_topic" value="/camera/depth/camera_info"/>
        <arg name="rtabmapviz" value="false"/>
        <arg name="rviz" value="true"/>

    </include>`
    
    I think something with node, I need to use this node? Because this `queue_size is 20` but when I launch qeue_size stay on 10.
    So same stuff with Grid/RayTracing.

Hagonata, 2023-08-21:
> Under rtabmap node:
> 
> ```
> <param name="Grid/RayTracing" type="string" value="true"/>
> ```

`    <include file="$(find rtabmap_ros)/launch/rtabmap.launch">
        <param name="queue_size" value="20"/>
        <param name="Grid/RayTracing" type="string" value="true"/>
        <arg name="args" value="--delete_db_on_start"/>
        <arg name="rgb_topic" value="/camera/color/image_raw"/>
        <arg name="depth_topic" value="/camera/aligned_depth_to_color/image_raw"/>
        <arg name="camera_info_topic" value="/camera/color/camera_info"/>
        <arg name="depth_camera_info_topic" value="/camera/depth/camera_info"/>
        <arg name="rtabmapviz" value="false"/>
        <arg name="rviz" value="true"/>

    </include>`
    
    I'm try to it already, but I think smth with my node. Because then I launch this I got `Grid/RayTracing = False` also same staff with a queue_size. I put value 20, but on the launch value is 10.

Hagonata, 2023-08-21:
> Under rtabmap node:
> 
> ```
> <param name="Grid/RayTracing" type="string" value="true"/>
> ```

I'm trying to do it already, but I think smth with my node. Because when I launched this I got `Grid/RayTracing = False` also the same staff with a queue_size. I put a value of 20, but on the launch value is 10.

 `    <include file="$(find rtabmap_ros)/launch/rtabmap.launch">
        <param name="queue_size" value="20"/>
        <param name="Grid/RayTracing" type="string" value="true"/>
        <arg name="args" value="--delete_db_on_start"/>
        <arg name="rgb_topic" value="/camera/color/image_raw"/>
        <arg name="depth_topic" value="/camera/aligned_depth_to_color/image_raw"/>
        <arg name="camera_info_topic" value="/camera/color/camera_info"/>
        <arg name="depth_camera_info_topic" value="/camera/depth/camera_info"/>
        <arg name="rtabmapviz" value="false"/>
        <arg name="rviz" value="true"/>

    </include>`

Hagonata, 2023-08-21:
> Under rtabmap node:
> 
> ```
> <param name="Grid/RayTracing" type="string" value="true"/>
> ```

I'm trying to do it already, but I think smth with my node. Because when I launched this I got Grid/RayTracing = False also the same staff with a queue_size. I put a value of 20, but on the launch value is 10.

```    
<include file="$(find rtabmap_ros)/launch/rtabmap.launch">
        <param name="queue_size" value="20"/>
        <param name="Grid/RayTracing" type="string" value="true"/>
        <arg name="args" value="--delete_db_on_start"/>
        <arg name="rgb_topic" value="/camera/color/image_raw"/>
        <arg name="depth_topic" value="/camera/aligned_depth_to_color/image_raw"/>
        <arg name="camera_info_topic" value="/camera/color/camera_info"/>
        <arg name="depth_camera_info_topic" value="/camera/depth/camera_info"/>
        <arg name="rtabmapviz" value="false"/>
        <arg name="rviz" value="true"/>

    </include> 
```

matlabbe, 2023-08-27:
With `rtabmap.launch`, you should use only `arg` (and for rtabmap library params, set them in args like below):
```xml
<arg name="queue_size" value="20"/>
<arg name="args" value="--delete_db_on_start --Grid/RayTracing true"/>
```

Hagonata, 2023-08-28:
> With `rtabmap.launch`, you should use only `arg` (and for rtabmap library params, set them in args like below):
> 
> ```
> <arg name="queue_size" value="20"/>
> <arg name="args" value="--delete_db_on_start --Grid/RayTracing true"/>
> ```

Thanks, will try it!