introlab/rtabmap #1104: RTABMAP with orbslam3

TakahikoHasegawa03, 2023-08-02:
Hi there,

im trying to use ORB_SLAM3 for visual odometry in rtabmap but I’m struggling to set the environment.

i did the following procedure:

1. Install ORBSLAM3 
git clone -b v0.4-beta https://github.com/UZ-SLAMLab/ORB_SLAM3.git
cd ORB_SLAM3/ 
wget https://gist.githubusercontent.com/matlabbe/f5cb281304a1305b2824a6ce19792e13/raw/f8bbc796edc29b9f815cbf3c99a0c3e13e23663d/orbslam3_v4_rtabmap_fix.patch
git apply orbslam3_v4_rtabmap_fix.patch 
./build.sh 
2. Configure the environment 
sudo gedit ~/.bashrc 
export ORB_SLAM_ROOT_DIR=/home/lib/ORB_SLAM3
source ~/.bashrc 
3. Build rtabmap in my work space
cd ~/home/ros2_ws/src/rtabmap
cd build/ 
cmake -DWITH_G2O=OFF -DWITH_ORB_SLAM=ON ..
make -j4
Make install
4. Build ros2 rtabmap 
cd ros2_ws/
Colcon build —packages-select rtabmap rtabmap_ros —cmake-args -DWITH_ORB_SLAM=ON

5. Source ros2_ws/install/setup.bash
6. Ros2 run rtabmap_ros rtabmap —version 
7. 

Then it still shows orbslam being false.

what am I doing wrong? 

Thanks in advance

matlabbe, 2023-08-06:
on step 3, after `cmake -DWITH_G2O=OFF -DWITH_ORB_SLAM=ON ..`, make sure you see in cmake info at the end that WITH_ORB_SLAM is YES. If so maybe the rtabmap_ros is linking on another installed version of rtabmap, make sure rtabmap binaries are not installed.

TakahikoHasegawa03, 2023-08-06:
Ok I have managed to get the environment set up now! 
but I have a few questions, can I use IMU input to make the odometry VIO? 

what’s the launch file for orbslam 3 like?

takahiko-hasegawa-hdjp, 2023-08-07:
Great thanks helped a lot!

fbliman, 2023-08-16:
Is there any docker file with orbslam integration working?

matlabbe, 2023-08-20:
There's one with ORB_SLAM2 here: https://github.com/introlab/rtabmap/blob/61dbe51ee3cea7a661d5a7a90c20a828074bd480/docker/jfr2018/latest/Dockerfile#L156-L158

matlabbe, 2023-08-27:
You may not have to apply the opencv3.1 patch for opencv 3.2.

zsbztdxx, 2023-09-23:
> Ok I have managed to get the environment set up now! but I have a few questions, can I use IMU input to make the odometry VIO?
> 
> what’s the launch file for orbslam 3 like?

Hi，I meet the same trouble as you showed at step 7，how did you solve it？

zsbztdxx, 2023-09-27:
> > Ok I have managed to get the environment set up now! but I have a few questions, can I use IMU input to make the odometry VIO?
> > what’s the launch file for orbslam 3 like?
> 
> Hi，I meet the same trouble as you showed at step 7，how did you solve it？

ok，I have solved it，just take rtabmap out of ros workspace，and set RTABMap_DIR in .bashrc