introlab/rtabmap #1106: Missing VINS-Support

den-schmidt, 2023-08-03:
Hello @matlabbe ,

I want to try out VINS-Fusion support for RTAB-Map. I have the following setup and status:

1. Docker Container with Ubuntu 18.04 & ROS Melodic
2. Installed all dependencies for VINS and MSCKF
3. Built the catkin workspace including rtabmap_ros, vins, and msckf_vio
4. I applied the VINS patch from here https://github.com/introlab/rtabmap/issues/380
5. I applied the MSCKF patch from here https://github.com/introlab/rtabmap/issues/304
6. Then, I moved to building rtabmap from source while setting the flags for VINS and MSCKF
7. Everything builds like a charm
8. Starting the following launch file tells me that I do not have VINS-Support...

launch file:

```
<!-- https://wiki.ros.org/rtabmap_ros/Tutorials/StereoHandHeldMapping -->

<launch>
	<!-- Parameters -->
    <arg name="online" default="false" />

    <!-- Start the RealSense camera -->
    <group if="$(arg online)">
        <node pkg="realsense2_camera" type="rs_camera.launch" name="realsense_camera" output="screen">
            <param name="enable_infra1" value="true" />
            <param name="enable_infra2" value="true" />
            <param name="unite_imu_method" value="linear_interpolation" /> 
			<!-- <param name="unite_imu_method" value="copy" /> -->
            <param name="enable_gyro" value="true" />
            <param name="enable_accel" value="true" />
            <param name="enable_sync" value="true" />
            <param name="initial_reset" value="true" />
            <param name="infra_width" value="1280" />
            <param name="infra_height" value="800" />
            <param name="infra_fps" value="30" />
            <param name="enable_color" value="false" />
            <param name="enable_depth" value="false" />
            <param name="json_file_path" value="/workspaces/vdi-cup/mxck_ws/src/slam/config/stereo_1280x800_30fps_emitter_disabled.json" />
        </node>
    </group>

	<!-- Start rosbag if parameter "online" is false -->
    <group unless="$(arg online)">
        <node pkg="rosbag" type="play" name="rosbag_player" args="$(find as_slam)/bagfiles/100723_stereo.bag" />
    </group>

    <!-- Start IMU filter -->
    <node pkg="imu_filter_madgwick" type="imu_filter_node" name="imu_filter" output="screen">
        <param name="use_mag" value="false" />
        <param name="publish_tf" value="false" />
        <param name="world_frame" value="enu" />
        <remap from="/imu/data_raw" to="/camera/imu" />
        <remap from="/imu/data" to="/rtabmap/imu" />
    </node>

    <!-- Start RTAB-Map -->
    <include file="$(find rtabmap_ros)/launch/rtabmap.launch">
        <arg name="rtabmap_args" value="--delete_db_on_start
                                        --Optimizer/GravitySigma 0.3
										--Odom/Strategy 9
                                        --OdomVINS/ConfigPath $(find vins)/../config/realsense_d435i/realsense_stereo_imu_config.yaml
										--Reg/Force3DoF false" />

		<arg name="left_image_topic" value="/camera/infra1/image_rect_raw" />
        <arg name="right_image_topic" value="/camera/infra2/image_rect_raw" />
        <arg name="left_camera_info_topic" value="/camera/infra1/camera_info" />
        <arg name="right_camera_info_topic" value="/camera/infra2/camera_info" />
        
		<arg name="stereo" value="true" />
[... 583 lines cut ...]
--   With ZEDOC                = NO (ZED Open Capture not found)
--   With RealSense            = NO (librealsense not found)
--   With RealSense2           = YES (License: Apache-2)
--   With MyntEyeS             = NO (mynteye s sdk not found)
--   With DepthAI              = NO (WITH_DEPTHAI=OFF)
-- 
--  Odometry Approaches:
--   With loam_velodyne        = NO (WITH_LOAM=OFF)
--   With floam                = NO (WITH_FLOAM=OFF)
--   With libfovis             = NO (WITH_FOVIS=OFF)
--   With libviso2             = NO (WITH_VISO2=OFF)
--   With dvo_core             = NO (WITH_DVO=OFF)
--   With okvis                = NO (okvis not found)
--   With msckf_vio            = YES (License: Penn Software License)
--   With VINS-Fusion          = YES (License: GPLv3)
--   With OpenVINS             = NO (WITH_OPENVINS=OFF)
--   With ORB_SLAM             = NO (WITH_ORB_SLAM=OFF)
-- Show all options with: cmake -LA | grep WITH_
-- --------------------------------------------
-- Configuring done
-- Generating done
-- Build files have been written to: /libraries/rtabmap/build`
```

Would you happen to know if I am missing something? 😄

den-schmidt, 2023-08-05:
The Docker file mentioned in this issue (https://github.com/introlab/rtabmap/issues/855) helped me because it states that the ```ldconfig``` command should be called after building to update the shared library cache of my system. Thank you anyway @matlabbe and keep up the good work!😃

closed 2023-08-05 by den-schmidt

matlabbe, 2023-08-06:
Maybe you have the ros rtabmap binaries also installed at the same time, and when building rtabmap_ros, it was linking on the ros installed version, not the one built from source. The binaries would have to be uninstalled to make sure everything is linking on right version of libraries on runtime the right libraries are used.