introlab/rtabmap #1110: RTabmap DB doesn't save optimized pose but only odometry pose

guoqingzh, 2023-08-10:
Hi All, 

I noticed that Rtabmap DB doesn't have optimized pose saved but only odometry pose, sensor data (RGB, depth) and loop closure constraints. It seems to do optimization on the fly when the database is opened. 

@matlabbe Is this the case ? Can you confirm ?

matlabbe, 2023-08-13:
Yes most of time, though for different purposes, we also keep the optimized graph under Admin table. https://github.com/introlab/rtabmap/blob/6d36fc1fcd8e1f6855d245cbe8b104ba0b05516e/corelib/src/resources/DatabaseSchema.sql.in#L123-L124

hellovuong, 2023-09-01:
Wow, That is a surprise, I also noticed that rtabmap optimizes the pose during initializing. Can you explain why rtabmap does that instead of saving optimized poses? By saving opt poses, will save init time and give more deterministic performance?

matlabbe, 2023-09-03:
It uses the optimized on initialization, that one of the purposes. After that there will be a reoptimization from those poses as guess if there is a loop closure or other events triggering a reoptimization.

arjay55, 2023-10-24:
Hi. I am doing post processing to the saved database map. The poses are better after post-processing. However when viewed using rtabmap GUI, the occupancy grid map looks like the original unprocessed map, but when opened using rtabmap-databaseViewer, the occupancy grid map is updated/fixed. Is the difference related to this issue?

matlabbe, 2023-10-25:
From the databaseViewer, you may have to trigger a new optimized map. File->Regenerate Optimizied 2d map...

arjay55, 2023-10-27:
Hello.

I tried it already, but still the old occupancy grid persist. There is no save button in rtabmap-databaseViewer right? So I just exit the program and the database should be overwritten right? I am using ros2 humble. Thanks.

arjay55, 2023-10-28:
Found a solution. Upon loading the db file in RTABMAP, I have to choose "Local Map Optimized" to show the grid same as in databaseViewer.

closed 2023-10-29 by matlabbe

kameel2311, 2026-03-31:
Hello ! I am loading the database on an edge compute and it is rather taking a while for the optimization to be re-run again - is there a way to actually save the database with optimized poses directly and skip that part after first mapping session ?

matlabbe, 2026-06-15:
Doing this:
```
_optimizedPoses.insert({retrievedId, s->getPose()});
```
even if `s->getPose()` is not odometry pose, but some optimized pose from different session, the graph optimization can move that pose anytime. 

The key would be:
> If you want to make sure that all original poses don't move, you may add a Prior link on each node, with optimized pose and very low covariance.

Instead of changing the `Node.pose` data in the database, add a Prior constraint (`Link` table) to that node using the optimized pose you want. On graph optimization, the optimizer (with `Optimizer/PriorsIgnored=false`) would keep that pose at that prior pose.