introlab/rtabmap #1111: [BUG] KdTreeMatcherInstensity when knn>1

dagata-mining, 2023-08-11:
At line
https://github.com/introlab/rtabmap/blob/f0ba56bf5e771fd0b0690f6a1b3cb10e67cbb5f7/corelib/src/icp/libpointmatcher.h#L518
When the knn is higher than 1, the matches.ids.coef sometimes equals -1 and create a fault

dagata-mining, 2023-08-11:
Corrected the bug using this approach:

```cpp
 if(knn > 1)
        {
            Matches matchesOrderedByIntensity(
                typename Matches::Dists(1, pointsCount),
                typename Matches::Ids(1, pointsCount)
            );
#pragma omp parallel for
            for (int i = 0; i < pointsCount; ++i)
            {
                float minDistance = std::numeric_limits<float>::max();
                bool minDistFound = false;
                for(int k=0; k<knn && k<filteredReferenceIntensity.rows(); ++k)
                {
                    int matchesIdsCoeff = matches.ids.coeff(k, i);
                    if (matchesIdsCoeff!=-1)
                    {
                        float distIntensity = fabs(
                            filteredReadingIntensity(0, i) - filteredReferenceIntensity(0, matchesIdsCoeff));
                        if (distIntensity < minDistance)
                        {
                            matchesOrderedByIntensity.ids.coeffRef(0, i) = matches.ids.coeff(k, i);
                            matchesOrderedByIntensity.dists.coeffRef(0, i) = matches.dists.coeff(k, i);
                            minDistance = distIntensity;
                            minDistFound = true;
                        }
                    }
                }
                if (!minDistFound)
                {
                    matchesOrderedByIntensity.ids.coeffRef(0, i) = matches.ids.coeff(0, i);
                    matchesOrderedByIntensity.dists.coeffRef(0, i) = matches.dists.coeff(0, i);
                }
            }
            matches = matchesOrderedByIntensity;
        }
```

matlabbe, 2023-08-13:
Added your changes in commit above.

Merci Alexandre!

closed 2023-08-13 by matlabbe