introlab/rtabmap #1113: RGB for PyDetector

mattiasmar, 2023-08-14:
Hello,
Is there some parameter that controls whether the PyDetector receives gray images or rgb images?

If not, would introducing a parameter that controls all the calls to cv::cvtColor(imageFrom, tmp, cv::COLOR_BGR2GRAY); in corelib/src/RegistrationVis.cpp be a complete (correct and enough) solution?

matlabbe, 2023-08-14:
Oh yeah, it is a good point if the external detector needs RGB. Ideally, we could make an abstract function (e.g., `isColorSupported()`) for `Features2D` class that would return if the detector supports RGB or not, then in `RegistrationVis`, if the detector doesn't support RGB, we don't call `cvtColor`. The `PyDetector` class could return true for that function, all other detectors would return false by default.

We would need to deal with asserts like [here](https://github.com/introlab/rtabmap/blob/cdd8cd9e34751a9d2bcbb98854efa9f16a18354c/corelib/src/Features2d.cpp#L673) and [here](https://github.com/introlab/rtabmap/blob/cdd8cd9e34751a9d2bcbb98854efa9f16a18354c/corelib/src/python/PyDetector.cpp#L93) depending if the detector supports RGB or not.

matlabbe, 2023-12-10:
I'll keep this open as it could be a nice enhancement.