introlab/rtabmap #1118: post processing - global iterative and 3d

fbliman, 2023-08-21:
Hi,

I have a couple of questions regarding post procesing.

-with respec to to the graph view, it seems that I am getting the whole graph, (if I set global iterative to 0) but I cannot get to merge one region that is duplicated and with it respecting loop closures (right down) 
-what is the slider in global iterative? and what are the numbers at the bottom (3696, 0, 1396, 102, 0, 922, 0, 0, 2242)
- when I choose the optmizar and stretegy, that only aplies to detecting more loop closures or how do I do a global optimization?
thanks
[image]

also when trying to recreate the 3d map the database viewer crashes, which parameters can I modify to make it lighter

matlabbe, 2023-08-27:
- Iteration 0 is without loop closure optimization (pure odometry), this is the number of optimization iterations.
- The numbers at the bottom are the number of type of links (N=neighbor or odometry contraints, G is global loop closures, LS is local loop closure in space, U is user added loop closures, GR is gravity constraints).

When you change optimizer parameters, click 2 times on "span to all maps" checkbox to trigger a new optimization using current parameters.

Increase Decimation/Voxel size when exporting, set normals to 0 to avoid normal computation.

fbliman, 2023-08-28:
Thanks!

how can I add "U is user added loop closures"?

matlabbe, 2023-08-29:
Each time in "Constraints view" you add a new loop closure, it will be a "U". If you do "Detect more loop closures", these loop closures will be "U" too.