introlab/rtabmap #1120: Assertion failed

MarcoPicciau, 2023-08-29:
I recompiled RTABMap for windows. Result look like correct and the RTabMap.exe start, but crash whene I press play button, to acquire from my realsense d400 camera.

this is the message that i see:
OpenCV(4.7.0) 
Error: Assertion failed ((unsigned)i1 < (unsigned)size.p[1]) 
in cv::Mat_<double>::operator (), file G:\stonex.slam\opencv\include\opencv2/core/mat.inl.hpp, line 1620

I understood that the problem is in: IMU.cpp
in this function:
void IMU::convertToBaseFrame()
{

---->  linearAcceleration_ = cv::Vec3d(v(0,0), v(0,1), v(0,2));

program crash on access "v" V(0,1)
if i try to swap the index and the problem doesn't occur

I don't undestand how solve this, probably is a compliler's define wrang.

Please some one can help me...

matlabbe, 2023-08-31:
Did you build with vcpkg?

MarcoPicciau, 2023-08-31:
I have built PCL, OpenCV, and G20 using CMake and Visual C++ 2019 in DEBUG mode.

Then, I compiled the RTabMap project on QT 11.0.2 in DEBUG mode.

In practice, there are some OpenCV asserts that seem to indicate access violations due to indices out of the correct range.

I recompiled OpenCV again in debug mode, but disabled the "CV_DbgAssert" macro that triggers the THROW, and the software works.

Best regards,
Marco Picciau