introlab/rtabmap #1122: Build is broke with ZED SDK v4.

luvwinnie, 2023-08-30:
I have this errors when building with the latest ZED SDK v4, Is there any plan to support v4?

```
--- stderr: rtabmap
/workspaces/isaac_ros-dev/src/rtabmap/corelib/src/camera/CameraStereoZed.cpp: In constructor ‘rtabmap::CameraStereoZed::CameraStereoZed(int, int, int, int, int, bool, float, const rtabmap::Transform&, bool, bool, int)’:
/workspaces/isaac_ros-dev/src/rtabmap/corelib/src/camera/CameraStereoZed.cpp:285:9: error: ‘SENSING_MODE’ is not a member of ‘sl’
  285 |     sl::SENSING_MODE sens = static_cast<sl::SENSING_MODE>(sensingMode_);
      |         ^~~~~~~~~~~~
In file included from /workspaces/isaac_ros-dev/src/rtabmap/corelib/src/../include/rtabmap/core/Landmark.h:33,
                 from /workspaces/isaac_ros-dev/src/rtabmap/corelib/src/../include/rtabmap/core/SensorData.h:42,
                 from /workspaces/isaac_ros-dev/src/rtabmap/corelib/src/../include/rtabmap/core/Camera.h:33,
                 from /workspaces/isaac_ros-dev/src/rtabmap/corelib/src/../include/rtabmap/core/camera/CameraVideo.h:31,
                 from /workspaces/isaac_ros-dev/src/rtabmap/corelib/src/../include/rtabmap/core/camera/CameraStereoZed.h:31,
                 from /workspaces/isaac_ros-dev/src/rtabmap/corelib/src/camera/CameraStereoZed.cpp:28:
/workspaces/isaac_ros-dev/src/rtabmap/corelib/src/camera/CameraStereoZed.cpp:289:13: error: ‘sens’ was not declared in this scope; did you mean ‘setns’?
  289 |     UASSERT(sens >= sl::SENSING_MODE::STANDARD && sens < sl::SENSING_MODE::LAST);
      |             ^~~~
/workspaces/isaac_ros-dev/src/rtabmap/utilite/src/../include/rtabmap/utilite/ULogger.h:66:33: note: in definition of macro ‘UASSERT’
   66 | #define UASSERT(condition) if(!(condition)) ULogger::write(ULogger::kFatal, __FILE__, __LINE__, __FUNCTION__, "Condition (%s) not met!", #condition)
      |                                 ^~~~~~~~~
/workspaces/isaac_ros-dev/src/rtabmap/corelib/src/camera/CameraStereoZed.cpp:289:25: error: ‘sl::SENSING_MODE’ has not been declared
  289 |     UASSERT(sens >= sl::SENSING_MODE::STANDARD && sens < sl::SENSING_MODE::LAST);
      |                         ^~~~~~~~~~~~
/workspaces/isaac_ros-dev/src/rtabmap/utilite/src/../include/rtabmap/utilite/ULogger.h:66:33: note: in definition of macro ‘UASSERT’
   66 | #define UASSERT(condition) if(!(condition)) ULogger::write(ULogger::kFatal, __FILE__, __LINE__, __FUNCTION__, "Condition (%s) not met!", #condition)
      |                                 ^~~~~~~~~
/workspaces/isaac_ros-dev/src/rtabmap/corelib/src/camera/CameraStereoZed.cpp:289:62: error: ‘sl::SENSING_MODE’ has not been declared
  289 |     UASSERT(sens >= sl::SENSING_MODE::STANDARD && sens < sl::SENSING_MODE::LAST);
      |                                                              ^~~~~~~~~~~~
/workspaces/isaac_ros-dev/src/rtabmap/utilite/src/../include/rtabmap/utilite/ULogger.h:66:33: note: in definition of macro ‘UASSERT’
   66 | #define UASSERT(condition) if(!(condition)) ULogger::write(ULogger::kFatal, __FILE__, __LINE__, __FUNCTION__, "Condition (%s) not met!", #condition)
      |                                 ^~~~~~~~~
/workspaces/isaac_ros-dev/src/rtabmap/corelib/src/camera/CameraStereoZed.cpp: In constructor ‘rtabmap::CameraStereoZed::CameraStereoZed(const string&, int, int, int, bool, float, const rtabmap::Transform&, bool, bool, int)’:
/workspaces/isaac_ros-dev/src/rtabmap/corelib/src/camera/CameraStereoZed.cpp:337:9: error: ‘SENSING_MODE’ is not a member of ‘sl’
  337 |     sl::SENSING_MODE sens = static_cast<sl::SENSING_MODE>(sensingMode_);
      |         ^~~~~~~~~~~~
In file included from /workspaces/isaac_ros-dev/src/rtabmap/corelib/src/../include/rtabmap/core/Landmark.h:33,
                 from /workspaces/isaac_ros-dev/src/rtabmap/corelib/src/../include/rtabmap/core/SensorData.h:42,
                 from /workspaces/isaac_ros-dev/src/rtabmap/corelib/src/../include/rtabmap/core/Camera.h:33,
                 from /workspaces/isaac_ros-dev/src/rtabmap/corelib/src/../include/rtabmap/core/camera/CameraVideo.h:31,
                 from /workspaces/isaac_ros-dev/src/rtabmap/corelib/src/../include/rtabmap/core/camera/CameraStereoZed.h:31,
                 from /workspaces/isaac_ros-dev/src/rtabmap/corelib/src/camera/CameraStereoZed.cpp:28:
/workspaces/isaac_ros-dev/src/rtabmap/corelib/src/camera/CameraStereoZed.cpp:341:13: error: ‘sens’ was not declared in this scope; did you mean ‘setns’?
  341 |     UASSERT(sens >= sl::SENSING_MODE::STANDARD && sens < sl::SENSING_MODE::LAST);
      |             ^~~~
/workspaces/isaac_ros-dev/src/rtabmap/utilite/src/../include/rtabmap/utilite/ULogger.h:66:33: note: in definition of macro ‘UASSERT’
   66 | #define UASSERT(condition) if(!(condition)) ULogger::write(ULogger::kFatal, __FILE__, __LINE__, __FUNCTION__, "Condition (%s) not met!", #condition)
      |                                 ^~~~~~~~~
/workspaces/isaac_ros-dev/src/rtabmap/corelib/src/camera/CameraStereoZed.cpp:341:25: error: ‘sl::SENSING_MODE’ has not been declared
  341 |     UASSERT(sens >= sl::SENSING_MODE::STANDARD && sens < sl::SENSING_MODE::LAST);
      |                         ^~~~~~~~~~~~
/workspaces/isaac_ros-dev/src/rtabmap/utilite/src/../include/rtabmap/utilite/ULogger.h:66:33: note: in definition of macro ‘UASSERT’
   66 | #define UASSERT(condition) if(!(condition)) ULogger::write(ULogger::kFatal, __FILE__, __LINE__, __FUNCTION__, "Condition (%s) not met!", #condition)
      |                                 ^~~~~~~~~
/workspaces/isaac_ros-dev/src/rtabmap/corelib/src/camera/CameraStereoZed.cpp:341:62: error: ‘sl::SENSING_MODE’ has not been declared
  341 |     UASSERT(sens >= sl::SENSING_MODE::STANDARD && sens < sl::SENSING_MODE::LAST);
      |                                                              ^~~~~~~~~~~~
/workspaces/isaac_ros-dev/src/rtabmap/utilite/src/../include/rtabmap/utilite/ULogger.h:66:33: note: in definition of macro ‘UASSERT’
   66 | #define UASSERT(condition) if(!(condition)) ULogger::write(ULogger::kFatal, __FILE__, __LINE__, __FUNCTION__, "Condition (%s) not met!", #condition)
      |                                 ^~~~~~~~~
/workspaces/isaac_ros-dev/src/rtabmap/corelib/src/camera/CameraStereoZed.cpp: In member function ‘virtual bool rtabmap::CameraStereoZed::init(const string&, const string&)’:
/workspaces/isaac_ros-dev/src/rtabmap/corelib/src/camera/CameraStereoZed.cpp:468:52: error: ‘struct sl::CameraInformation’ has no member named ‘calibration_parameters’
  468 |  sl::CalibrationParameters *stereoParams = &(infos.calibration_parameters );
      |                                                    ^~~~~~~~~~~~~~~~~~~~~~
/workspaces/isaac_ros-dev/src/rtabmap/corelib/src/camera/CameraStereoZed.cpp:476:17: error: ‘struct sl::CalibrationParameters’ has no member named ‘T’
  476 |   stereoParams->T[0],//baseline
      |                 ^
In file included from /workspaces/isaac_ros-dev/src/rtabmap/corelib/src/../include/rtabmap/core/Landmark.h:33,
                 from /workspaces/isaac_ros-dev/src/rtabmap/corelib/src/../include/rtabmap/core/SensorData.h:42,
                 from /workspaces/isaac_ros-dev/src/rtabmap/corelib/src/../include/rtabmap/core/Camera.h:33,
[... 39 lines cut ...]
      |                                       ^~~~~~~~~~~~
make[2]: *** [corelib/src/CMakeFiles/rtabmap_core.dir/build.make:468: corelib/src/CMakeFiles/rtabmap_core.dir/camera/CameraStereoZed.cpp.o] Error 1
make[2]: *** Waiting for unfinished jobs....
make[1]: *** [CMakeFiles/Makefile2:826: corelib/src/CMakeFiles/rtabmap_core.dir/all] Error 2
make: *** [Makefile:156: all] Error 2
---
Failed   <<< rtabmap [35.4s, exited with code 2]
Aborted  <<< velodyne_pointcloud [3.33s]
Aborted  <<< rtcm_msgs [0.15s]
Aborted  <<< marti_common_msgs [21.4s]
Aborted  <<< joint_state_publisher [3.10s]
Aborted  <<< control_toolbox [1.44s]
Aborted  <<< hardware_interface [2.25s]
Aborted  <<< zed_interfaces [35.6s]
Aborted  <<< rtabmap_msgs [36.8s]
Aborted  <<< geographic_msgs [37.5s]
Aborted  <<< gazebo_msgs [14.4s]
Aborted  <<< rviz_rendering [1min 11s]

Summary: 38 packages finished [1min 19s]
  1 package failed: rtabmap
  11 packages aborted: control_toolbox gazebo_msgs geographic_msgs hardware_interface joint_state_publisher marti_common_msgs rtabmap_msgs rtcm_msgs rviz_rendering velodyne_pointcloud zed_interfaces
  1 package had stderr output: rtabmap
  137 packages not processed
```

matlabbe, 2023-08-31:
Oh well, there is a plan now. I'll take a look in the next days. Thx for reporting.

matlabbe, 2023-09-04:
Which version are you running. Just tried and it worked, then looking at the code, the update was done 5 months ago https://github.com/introlab/rtabmap/commit/e54195c47f4de8d94b99bb7308f1c45e4eeb206d

luvwinnie, 2023-09-05:
I'm using v4.0.3 SDK, i tried again with humble-devel branch it shows the same errors.

matlabbe, 2023-09-05:
Try master branch to get latest changes and use ros2 branch on rtabmap_ros

luvwinnie, 2023-09-09:
@matlabbe Thank you! I'm able to build with the master branch.

closed 2023-09-09 by luvwinnie