introlab/rtabmap #1127: [Feature-Request] ANMS

borongyuan, 2023-09-03:
This [issue](https://github.com/introlab/rtabmap/pull/1058#issuecomment-1600053940) encountered when running SuperPoint on OAK should be related to NMS implementation. When I was learning about [Kimera-VIO](https://github.com/MIT-SPARK/Kimera-VIO) recently I noticed their use of [ANMS](https://github.com/BAILOOL/ANMS-Codes). It may help improve post-processing of SuperPoint, and can be used for other types of keypoints as well.

matlabbe, 2023-09-04:
Having better distributed keypoints can increase robustness of visual odometry and better accuracy. The implementations seem not using the keypoint feature response to select "best" features (like the current [NMS](https://github.com/introlab/rtabmap/blob/e56e0c37f767987cd02407a0be28f0f40696768b/corelib/src/util2d.cpp#L2096-L2205) implementation for superpoint), they seem only using 2d position of each feature and remove them uniformly. These algorithms could be useful for other detectors, and they could be used by [Feature2D::limitKeypoints()](https://github.com/introlab/rtabmap/blob/e56e0c37f767987cd02407a0be28f0f40696768b/corelib/src/Features2d.cpp#L283C17-L418), instead of just using keypoint's response value, a [ANMS](https://github.com/BAILOOL/ANMS-Codes) approach could be used instead to get a better distribution of features in the image.

In complementary, it would also be nice to integrate [Kimera-VIO](https://github.com/MIT-SPARK/Kimera-VIO) as built-in odomerty approach in rtabmap standalone, or having a ros example how to integrate their [ros package](https://github.com/MIT-SPARK/Kimera-VIO-ROS) with rtabmap_ros.

borongyuan, 2023-09-05:
Keypoint response is used. The algorithm will sort keypoints by strength at the beginning. But it's certainly not the only factor being considered. To obtain a more even distribution, keypoints with higher response may be filtered out. But SuperPoint itself also tends to generate evenly distributed keypoints. Therefore, I think it is possible to make them work together through appropriate parameter configuration or algorithm improvements.

Yep, it should work perfectly for [Feature2D::limitKeypoints()](https://github.com/introlab/rtabmap/blob/e56e0c37f767987cd02407a0be28f0f40696768b/corelib/src/Features2d.cpp#L283C17-L418).

closed 2024-05-20 by borongyuan