introlab/rtabmap #1128: [Feature-Request] BEBLID

borongyuan, 2023-09-03:
> BEBLID (Boosted Efficient Binary Local Image Descriptor) is a efficient binary descriptor learned with boosting. It is able to describe keypoints from any detector just by changing the scale_factor parameter. In several benchmarks it has proved to largely improve other binary descriptors like ORB or BRISK with the same efficiency.

BEBLID is one of the best non-deep learning feature descriptor in recent years. It was verified in BEBLID-SLAM. I think we should add it in and see the effect. What's a bit awkward is that [it has only been added to OpenCV since 4.5.1](https://docs.opencv.org/4.5.1/d7/d99/classcv_1_1xfeatures2d_1_1BEBLID.html). We have several platforms using slightly older versions of OpenCV. For older OpenCV versions, we can use the implementation in [this repository](https://github.com/iago-suarez/BEBLID). But I also want to avoid messing up the code too much.

matlabbe, 2023-09-04:
[BEBLID](https://github.com/iago-suarez/BEBLID) is BSD license, so we could probably include their source code as 3rd party in rtabmap and use it when we detect OpencV<4.5.1, and then use opencv's version when Opencv>=4.5.1 is detected. It seems to support many detectors, I think we could start by making a new Feature2D type for FAST+BEBLID (which seems the type used in [BEBLID-SLAM](https://ieeexplore.ieee.org/stamp/stamp.jsp?tp=&arnumber=10023629))

borongyuan, 2023-09-05:
> [BEBLID](https://github.com/iago-suarez/BEBLID) is BSD license, so we could probably include their source code as 3rd party in rtabmap and use it when we detect OpencV<4.5.1, and then use opencv's version when Opencv>=4.5.1 is detected.

I thought so at the beginning, but we may need to check whether there is any difference between the implementations to avoid parameter incompatibility. For example for scale_factor in [BEBLID](https://github.com/iago-suarez/BEBLID):
> 6.25f is default and fits for KAZE, SURF detected keypoints window ratio
> 6.75f should be the scale for SIFT detected keypoints window ratio
> 5.00f should be the scale for AKAZE, MSD, AGAST, FAST, BRISK keypoints window ratio
> 0.75f should be the scale for ORB keypoints ratio
> 1.50f was the default in original implementation

And in OpenCV's doc:
> 1.00f should be the scale for [ORB](https://docs.opencv.org/4.x/db/d95/classcv_1_1ORB.html) keypoints
> 6.75f should be the scale for [SIFT](https://docs.opencv.org/4.x/d7/d60/classcv_1_1SIFT.html) detected keypoints
> 6.25f is default and fits for [KAZE](https://docs.opencv.org/4.x/d3/d61/classcv_1_1KAZE.html), [SURF](https://docs.opencv.org/4.x/d5/df7/classcv_1_1xfeatures2d_1_1SURF.html) detected keypoints
> 5.00f should be the scale for [AKAZE](https://docs.opencv.org/4.x/d8/d30/classcv_1_1AKAZE.html), MSD, AGAST, FAST, [BRISK](https://docs.opencv.org/4.x/de/dbf/classcv_1_1BRISK.html) keypoints

I don't know why there is a difference for ORB.