introlab/rtabmap #1129: [Feature-Request] LightGlue

borongyuan, 2023-09-03:
I had some conversations with the OAK hardware team last week. Their subsequent Robotics Vision Core 4 (RVC4) will have great performance improvements (48 INT8, 12 FP16 TOPS). Therefore, we hope to deploy more neural networks on OAK in the future. For example, the newly released [LightGlue](https://github.com/cvg/LightGlue) this year already has the support of ONNX and Kornia.

cdb0y511, 2023-09-04:
This is related to [#1123](https://github.com/introlab/rtabmap/issues/1123). The simplest way to test this PyTorch-based matcher is via Python API like pydetect (which has superGlue enabled.) I hope @matlabbe can provide a pure Python API example or Python test file. I have difficulty developing a mixture of C++ and Python applications. The good news is the popular feature detecting superpoint seems embedded in C++. But I hope he can provide full Python detect API to embed recent features, like DISK.

cdb0y511, 2023-09-05:
I think the vis part (focus on visual registration) should be considered to give Python API first, similar to PyMatcher (https://github.com/introlab/rtabmap/blob/master/corelib/src/python/rtabmap_superglue.py, which has superGlue enabled.) And PyDetector (https://github.com/introlab/rtabmap/blob/master/corelib/src/python/rtabmap_superpoint.py). For the Python thread freezing issue, https://github.com/introlab/rtabmap/issues/896#issuecomment-1575742541 can be avoided via the docker environment. I think some documents about Python input and return (https://github.com/introlab/rtabmap/blob/master/corelib/src/python/rtabmap_superglue.py, https://github.com/introlab/rtabmap/blob/master/corelib/src/python/rtabmap_superpoint.py))  would be nice; this will save a lot of time for latecomers to print the data structure.

cdb0y511, 2023-11-07:
> DISK is also included in [kornia.feature](https://kornia.readthedocs.io/en/latest/feature.html). And kornia is based on pytorch. Some kornia modules can be exported to ONNX. So I'm more interested in supporting onnxruntime. This makes it easy to deploy the same model on different platforms and hardwares. PyTorch is generally mainly used for model development and training, and its inference performance is not outstanding.
Hi, @borongyuan, @matlabbe 
I noticed there is an ONNX version of lightGlue, https://github.com/OroChippw/LightGlue-OnnxRunner.
I think the C++ version is more suitable for rtabmap since the Python one always has some threading issues with me (segmentation fault with python thread sometimes).
Any plans to try https://github.com/OroChippw/LightGlue-OnnxRunner?
SuperGlue solving optimal transport problems with Sinkhorn is horribly slow.

borongyuan, 2023-11-20:
Currently only features detection and matching can use neural networks. I'm also looking at some new technologies, including NeRF and Gaussian Splatting. RTAB-Map should have the potential to support these new technologies. I had hoped to use ONNX to make using neural networks easier. But NeRF requires model training, so PyTorch may still be needed.