introlab/rtabmap #1133: Make error with Generating ui_preferencesDialog.h on Docker, Need help.

cdb0y511, 2023-09-12:
Hi, @matlabbe.
Typically, with apt install libpcl-dev on docker files (similar to https://github.com/introlab/rtabmap/tree/master/docker/frontiers2022 but with latest FROM nvcr.io/nvidia/pytorch:23.07-py3 (ubuntu 22.04) and some extra package like libpoitmatcher.), it comes with a related qt5 package installed, all build and run smoothly. However, the vtk9 continues to generate stupid warnings like (https://github.com/PointCloudLibrary/pcl/issues/4830). So, I have to build the PCL and vtk from the source with  VTK_LEGACY_REMOVE=ON to stop it.  

I manually apt install qtcreator qtbase5-dev qt5-qmake qttools5-private-dev qtdeclarative5-dev for qt. Rebuild PCL and OpenCV. However, when build rtabmap from source, Generating ui_preferencesDialog.h or ui_*
It outputs make error: /bin/bash: line 1: -o: command not found which locate to "cd /workspace/rtabmap/build/guilib/src && -o /workspace/rtabmap/build/guilib/src/ui_preferencesDialog.h" in make files. 
I highly suspect the qt or qtcreator-related problem may lack some qt dependency because it relates to ui_*.h.

Any idea to fix it?

[ 43%] Generating ui_preferencesDialog.h
**cd /workspace/rtabmap/build/guilib/src && -o /workspace/rtabmap/build/guilib/src/ui_preferencesDialog.h** /workspace/rtabmap/guilib/src/ui/preferencesDialog.ui
**/bin/bash: line 1: -o: command not found**
make  -f tools/StereoEval/CMakeFiles/stereoEval.dir/build.make tools/StereoEval/CMakeFiles/stereoEval.dir/build
**make[2]: *** [guilib/src/CMakeFiles/rtabmap_gui.dir/build.make:214: guilib/src/ui_preferencesDialog.h] Error 127**

cdb0y511, 2023-09-12:
I dont know witch qt package is missing. 
qmake -v
QMake version 3.1
Using Qt version 5.15.3 in /usr/lib/x86_64-linux-gnu

cdb0y511, 2023-09-14:
Hi, @matlabbe.I made a pr to fix this issue.

matlabbe, 2023-09-14:
PR is merged! thx

closed 2023-09-14 by matlabbe