introlab/rtabmap #1136: About RAE from DepthAI

borongyuan, 2023-09-15:
Hi @matlabbe ,
I just got the [RAE](https://docs.luxonis.com/projects/hardware/en/latest/pages/rae/) today. [It can also run RTAB-Map via ROS2.](https://github.com/luxonis/rae-ros) Have you also ordered this hardware? Or I can ask their team to see if they can send you one.
[image]

matlabbe, 2023-09-15:
Nice to see peiple starting to get them. I am also on the pre-orders list, I should receive it eventually!

borongyuan, 2023-09-21:
Great, they will keep an eye on your order and arrange shipping as soon as possible. But the current system is still very immature, and we are currently stuck in the first step of network connection. Everyone is still waiting for them to deal with it.

closed 2023-09-21 by borongyuan

matlabbe, 2023-09-21:
Thanks for the update, I think this launch can be improved to use both cameras for localization: https://github.com/luxonis/rae-ros/blob/humble/rae_bringup/launch/rtabmap.launch.py

Maybe adding an option for multi-camera VO too.