introlab/rtabmap #1138: iOS app architecture

wetoo-cando, 2023-09-23:
Great work @matlabbe  and thanks for making it available to the community. 

I am trying to build an iOS app that uses the color and laser data and in which most of the libraries are python/C++. Since rtabmap already essentially does that, it would help me a lot to understand how the iOS app works. Of course your open source code helps. But since I'm new to iOS development, I'm wondering: is there some high level documentation that describes  how the iOS app interacts with the main rtabmap C++ library. If not, could you please describe it here in a few sentences?

Thanks!

matlabbe, 2023-09-26:
There is no documentation about the Swift to C++ interface. Basically, swift calls Objective C functions that calls C functions. We pass the c++ rtabmap object as a pointer to C functions.

In summary the swift code is mainly UI stuff, calling the c++ library under the hood. For ARkit, it forwards the pose and image data to c++ library, which does the loop closure detection and the opengl rendering in c++.

We use the same opengl backed for the android and ios app.

wetoo-cando, 2023-09-26:
Thanks!

In the ios code, you include the RtabMapApp.h header. Is this the same as rtabmap/app/android/jni
/RTABMapApp.h?

Or is this a header that somehow gets automatically generated during the ios build?

matlabbe, 2023-10-01:
ios build is including `rtabmap/app/android/jni` folder. I didn't put time to move that code in a shared folder between android and ios.

wetoo-cando, 2023-10-07:
Ok thanks!

Gyudori, 2024-02-07:
Hello, I'm new to iOS app development, and I'm trying to understand the main pipeline and architecture of iOS RTAB-Map. 

From what I've gathered, it seems that iOS RTAB-Map doesn't primarily rely on core algorithms like odometry and loop closure from the RTAB-Map core. Is this correct? 
Instead, does the RTAB-Map app primarily leverage the outputs of ARKit for its functionality?

matlabbe, 2024-02-07:
RTABMap iOS app(without lidar) is using these inputs from ARKit: pose, RGB frame and 3D tracked features. Loop closure is still done by RTAB-Map library (we reproject the 3D tracked features in RGB frame to extract the corresponding visual descriptors), as well as the 3D mesh reconstruction/texturing.

Gyudori, 2024-02-08:
Thank you so much for the prompt response!!!
I will check based on the information you provided.

wetoo-cando, 2025-04-15:
> For ARkit, it forwards the pose and image data to c++ library, which does the loop closure detection and the opengl rendering in c++.
> 
> We use the same opengl backend for the android and ios app.

@matlabbe can you please elaborate on that? Where is this "same opengl backend" located?

wetoo-cando, 2025-04-15:
Ah sorry, you've already answered that above. I'm still unsure how the GL lib part gets built though.

matlabbe, 2025-04-19:
The iOS project contains directly references to opengl es code from [android/jni](https://github.com/introlab/rtabmap/tree/master/app/android/jni). All the opengl code is built inside the main ios c++ library of the App.