introlab/rtabmap #1139: Calculating localization accuracy

vishaldolase-honda, 2023-09-26:
Hi,@matlabbe,

I am using RBG-D data + Odomentry data
mapping and localization are working fine

i want to measure the quality of online localization
i have found covariance data from /rtabmap/localization_pose and word_keypoint, word_point and descriptor data from /rtabmap/mapData
i am using g2o for graph optimization and know the the covariance set in localization_pose topic is computed from the marginals after graph optimization.
also please share how we can utilize imu data here
please guide me for calculating confidence value to measure the quality of online localization.

matlabbe, 2023-10-01:
On localization, covariance is indeed based on output of g2o.

If there is no localization, the odometry covariance is propagated to previous localization covariance.

In term of localization quality, covariance is one way, but you can also check the number of visual inliers of a loop closure detection. In general, the more inliers are used to compute the localization transform, the better it will be.

vishaldolase-honda, 2023-10-02:
Understood, i will check the number of visual inliers.
also let me know if below values can be useful or not 
loopThr is our loop closure, and we compare it with _highestHypothesis, can i use these values also ?

[_highestHypothesis.second >= loopThr](https://github.com/introlab/rtabmap/blob/8d23f19f6c2d3ddb222d63aab2960aaa3df59de4/corelib/src/Rtabmap.cpp#L1974C5-L1974C45)

matlabbe, 2023-10-02:
You can increase `Rtabmap/LoopThr` parameter (default is `0.11`). Increasing that parameter will decrease false positives, but also decrease true positives. You can see Min TLoop for 100% precision in those various datasets in Table II of this paper [Appearance-Based Loop Closure Detection for Online Large-Scale and Long-Term Operation](https://introlab.3it.usherbrooke.ca/mediawiki-introlab/images/b/bc/TRO2013.pdf).

In `rtabmap/info` topic, you should be able to see all values of these statistics.

vishaldolase-honda, 2023-10-04:
Thank you very much :)

closed 2023-10-04 by matlabbe