introlab/rtabmap #1140: window build and meet exe error

wangjinhoon, 2023-09-27:
I followed the installation process on Windows and executed the bin\RTAMAP.exe file. However, I'm encountering this error. What could be wrong?

[image]

it's mean "The procedure entry point 'z_inflatelnit2_' could not be found in the DLL C:\Qt\6.2\msvc2019\bin\Qt6Svg.dll."

Is there any way to solve this?

help me, thank

matlabbe, 2023-10-01:
> I followed the installation process on Windows

Which one? do you use the binaries? compiling from source? On windows, we now recommend to build with vcpkg.

wangjinhoon, 2023-10-04:
Thank you for your response.
I was able to build the executable using vcpkg. 
However, I am having trouble configuring the parameters for the D435i. 
What could be the issue?
[image]
thank

matlabbe, 2023-10-04:
You would have to add [realsense2](https://github.com/microsoft/vcpkg/blob/accd79817981f1ce00c6a4164aecf2b2ac699823/ports/rtabmap/vcpkg.json#L117) option when configuring vcpkg build.

wangjinhoon, 2023-10-05:
Is this right?
```
vcpkg install rtabmap[realsense2] --triplet x64-windows --clean-after-build
```

i try it but i have same issue
thank

matlabbe, 2023-10-14:
Humm, just tried:
```
cd vcpkg
git pull
vcpkg install rtabmap[realsense2] --clean-after-build --recurse
```
and it seems okay:

[image]

wangjinhoon, 2023-10-15:
I couldn't solve this problem, but when I tried on a different computer, it worked. 

Thank you for your response!!

closed 2023-10-15 by matlabbe