introlab/rtabmap #1141: rtabmap matching with superglue

pxy522, 2023-10-02:
when i change the preferences -> motion estimation -> visual registration -> features Matching to Pymatcher, i try to use the rtabmap_superglue.py, but their effects are all the same no matter what script path i select, even the wrong script.

matlabbe, 2023-10-02:
Can you enable Debug log and show the log when a transformation is computed?

pxy522, 2023-10-21:
I appreciate your help, but I was able to solve the problem myself. It turned out to be a minor issue that I had overlooked due to a small detail. Thank you for your assistance！

mattiasmar, 2023-12-05:
Can one use pymatcher (e.g. SuperGlue) during localization, if KNN was used for mapping?

@pxy522 What did you try?

matlabbe, 2023-12-10:
> Can one use pymatcher (e.g. SuperGlue) during localization, if KNN was used for mapping?

Yes