introlab/rtabmap #1147: Cannot open database created by RTAB-MAP iOS when LiDAR Mode is false

takanotume24, 2023-10-20:
Hello.

In RTAB-MAP iOS, I scan in ARKit Mode (=`LiDAR Mode` set to False) and save the database.

Then when I try to open the database again, I get an error message `Failed to open database: Out of memory! Try again after lowering Point Cloud Density in Settings.` and cannot open the database.

I am trying this on an iPad Pro with an M1 chip, and a 1.3 GB database created with LiDAR Mode set to True opens fine. However, a 21MB database created in ARKit Mode gives the above error.

I followed the error message and set the `Settings` -> `Point Cloud Density` to `Very Low`, but still the same situation.

Is there any solution?

Thank you for all the support you provide.

matlabbe, 2023-10-21:
Currently reviewed on AppStore, the bug will be fixed in next released version (0.21.2)

closed 2023-10-21 by matlabbe

takanotume24, 2023-10-23:
Thank you for confirming this. I will check mine in the next release.

takanotume24, 2023-10-25:
The update has been delivered to my iPad and I have confirmed that I can open the database created in LiDAR Mode! Thanks again for your quick response!