introlab/rtabmap #1157: Manually set status in WM

MiuraShuhei, 2023-11-12:
Is there a way to manually set the status in the WM?
Operate in SLAM mode again from the origin on the map already created. I want to put only the nodes near the origin into the WM.

I would appreciate it if you could tell me if there is a better way.

matlabbe, 2023-11-18:
Is memory management enabled? If not, all nodes are in WM. Can you describe what issue you are trying to solve?

MiuraShuhei, 2023-11-19:
@matlabbe   Thank you for your response.

Memory management is enabled.

First, I map in SLAM mode. Next, I start localizing and updating the map from the origin coordinates in SLAM mode. This is because the environment is dynamic. At this time, I want to transfer only the nodes near the origin to WM memory. If InitWMWithAllNodes is enabled, it will take time until the proper number of nodes in WM is reached. If InitWMWithAllNodes is disabled, the last state is retained in WM, so in my case Not appropriate.

matlabbe, 2023-11-20:
The natural way to make sure that the origin (or any important location of interest) is staying in WM is to keep the robot still when mapping to build up the weight of the nodes added to map in that area. With high weight, they will stay longer in WM. If rtabmap_viz's loop closure view doesn't change to blue when not moving (indicating weight is increased for current node), maybe `Mem/RehearsalSimilarity` is too high.

To explicitly force rtabmap to transfer back nodes from LTM to WM on start, we would need to add a new parameter or service to tell rtabmap to load nodes around (inside a fixed radius) a node with specific "label" or "id".

Another hacky way would be to edit the [optimized local graph](https://github.com/introlab/rtabmap/blob/a1d43d23533e0846c1dd3d8eb264c54b63112ac9/corelib/src/resources/DatabaseSchema.sql.in#L123-L124) saved in the database before restarting another session.