introlab/rtabmap #1158: How to receive real-time odometry in Windows(OS)

wangjinhoon, 2023-11-13:
Hello, thank you for your help.
I want to receive real-time odometry estimated by RTAB-Map in Unity. Is there a way to integrate RTAB-Map with ROS2 on Windows or extract odometry in real-time?

If code modification is necessary, which parts should I look into?
Thank you. Have a great day."

matlabbe, 2023-11-18:
Rtabmap will publish `localization_pose` topic once every localization update (default 1Hz). If you need higher framerate, like at the same frame rate than odometry, you may subscribe to odometry topic and TF, then transform the odometry topic into map frame and republish it. You may also just use TF and [lookupTransform](http://docs.ros.org/en/noetic/api/tf/html/c++/classtf_1_1Transformer.html#ac59f0533c82c2ab9b834039d4cf3180b) between `map` and `odom.child_frame_id` frames when you receive a new odometry topic.

wangjinhoon, 2023-11-20:
Thank you for your response. I will revisit after examining the localization_pose once.

wangjinhoon, 2023-11-22:
If any issues arise, I will reopen it. Thank you.

closed 2023-11-22 by wangjinhoon