introlab/rtabmap #1166: Use RTAB-Map + OpenVINS with ZED2 Problem

ericlai0323, 2023-11-22:
I compiled OpenVINS and RTAB-Map from source, but I get warnings when running them saying my TF tree connection is not normal. Can someone please help me check what the issue might be? 

Thank you very much!!!

Below is the warning message:

```
[ INFO] [1700626779.170679455]: Odom: quality=0, std dev=0.000000m|0.000000rad, update time=0.003912s
[ INFO] [1700626779.323585268]: Odom: quality=0, std dev=0.000000m|0.000000rad, update time=0.000348s
[ INFO] [1700626779.366120078]: Odom: quality=0, std dev=0.000000m|0.000000rad, update time=0.000252s
[ INFO] [1700626779.430471599]: Odom: quality=0, std dev=0.000000m|0.000000rad, update time=0.000240s
[ INFO] [1700626779.494406700]: Odom: quality=0, std dev=0.000000m|0.000000rad, update time=0.000248s
[ INFO] [1700626779.561270333]: Odom: quality=0, std dev=0.000000m|0.000000rad, update time=0.000228s
[ INFO] [1700626779.629693937]: Odom: quality=0, std dev=0.000000m|0.000000rad, update time=0.000182s
[ WARN] [1700626779.678284194]: Could not get transform from odom to base_link after 0.500000 seconds (for stamp=1700626779.151464)! Error="Could not find a connection between 'odom' and 'base_link' because they are not part of the same tree.Tf has two or more unconnected trees.. canTransform returned after 0.503433 timeout was 0.5.".
[ INFO] [1700626779.700378208]: Odom: quality=0, std dev=0.000000m|0.000000rad, update time=0.000215s
[ INFO] [1700626779.765972340]: Odom: quality=0, std dev=0.000000m|0.000000rad, update time=0.000284s
[ INFO] [1700626779.831533127]: Odom: quality=0, std dev=0.000000m|0.000000rad, update time=0.000216s
[ INFO] [1700626779.897267018]: Odom: quality=0, std dev=0.000000m|0.000000rad, update time=0.000175s
[ INFO] [1700626779.962457252]: Odom: quality=0, std dev=0.000000m|0.000000rad, update time=0.000217s
[ INFO] [1700626780.031689608]: Odom: quality=0, std dev=0.000000m|0.000000rad, update time=0.000242s
[ INFO] [1700626780.096818767]: Odom: quality=0, std dev=0.000000m|0.000000rad, update time=0.000404s
[ INFO] [1700626780.162721815]: Odom: quality=0, std dev=0.000000m|0.000000rad, update time=0.000266s
[ WARN] [1700626780.184400238]: Could not get transform from odom to base_link after 0.500000 seconds (for stamp=1700626779.312418)! Error="Could not find a connection between 'odom' and 'base_link' because they are not part of the same tree.Tf has two or more unconnected trees.. canTransform returned after 0.502625 timeout was 0.5.".
[ INFO] [1700626780.229442245]: Odom: quality=0, std dev=0.000000m|0.000000rad, update time=0.000233s
[ INFO] [1700626780.296072302]: Odom: quality=0, std dev=0.000000m|0.000000rad, update time=0.000198s
[ INFO] [1700626780.366343862]: Odom: quality=0, std dev=0.000000m|0.000000rad, update time=0.000196s
[ INFO] [1700626780.434357990]: Odom: quality=0, std dev=0.000000m|0.000000rad, update time=0.000229s
[ INFO] [1700626780.498312811]: Odom: quality=0, std dev=0.000000m|0.000000rad, update time=0.000218s
[ INFO] [1700626780.564963036]: Odom: quality=0, std dev=0.000000m|0.000000rad, update time=0.000239s
[ INFO] [1700626780.630942210]: Odom: quality=0, std dev=0.000000m|0.000000rad, update time=0.000199s
[ WARN] [1700626780.689319134]: Could not get transform from odom to base_link after 0.500000 seconds (for stamp=1700626779.358052)! Error="Could not find a connection between 'odom' and 'base_link' because they are not part of the same tree.Tf has two or more unconnected trees.. canTransform returned after 0.503172 timeout was 0.5.".
```

Below is launch file:
```
<?xml version="1.0"?>

<launch>

  <!-- <node pkg="hector_trajectory_server" type="hector_trajectory_server" name="hector_trajectory_server" output="screen">
    <param name="target_frame_name" type="string" value="map" />
    <param name="source_frame_name" type="string" value="base_link" />
    <param name="trajectory_update_rate" type="double" value="4" />
    <param name="trajectory_publish_rate" type="double" value="0.25" />
  </node> -->

  <!-- Choose between depth and stereo, set both to false to do only scan -->
  <arg name="stereo"                    default="true"/>
  <arg     if="$(arg stereo)" name="depth"  default="false"/>
  <arg unless="$(arg stereo)" name="depth"  default="true"/>
  <arg name="subscribe_rgb"  default="$(arg depth)"/>

  <!-- Choose visualization -->
  <arg name="rtabmap_viz"             default="false" />
  <arg name="rviz"                    default="true" />

  <!-- Localization-only mode -->
  <arg name="localization"            default="false"/>
  <arg name="initial_pose"            default=""/>             <!-- Format: "x y z roll pitch yaw" or "x y z qx qy qz qw". Default: see "RGBD/StartAtOrigin" doc -->

  <!-- sim time for convenience, if playing a rosbag -->
  <arg name="use_sim_time"            default="false"/>
  <param if="$(arg use_sim_time)" name="use_sim_time" value="true"/>

  <!-- Corresponding config files -->
  <arg name="cfg"                     default="" /> <!-- To change RTAB-Map's parameters, set the path of config file (*.ini) generated by the standalone app -->
  <arg name="gui_cfg"                 default="~/.ros/rtabmap_gui.ini" />
  <arg name="rviz_cfg"                default="/home/ericlai/Launch/config/zed2_stereo.rviz" />

[... 481 lines cut ...]

  </group>

  <!-- Visualization RVIZ -->
  <node if="$(arg rviz)" pkg="rviz" type="rviz" name="rviz" args="-d $(arg rviz_cfg)"/>
  <node if="$(arg rviz)" pkg="nodelet" type="nodelet" name="points_xyzrgb" args="standalone rtabmap_util/point_cloud_xyzrgb" clear_params="$(arg clear_params)" output="$(arg output)">
    <remap if="$(arg stereo)" from="left/image"        to="$(arg left_image_topic_relay)"/>
    <remap if="$(arg stereo)" from="right/image"       to="$(arg right_image_topic_relay)"/>
    <remap if="$(arg stereo)" from="left/camera_info"  to="$(arg left_camera_info_topic)"/>
    <remap if="$(arg stereo)" from="right/camera_info" to="$(arg right_camera_info_topic)"/>
    <remap unless="$(arg subscribe_rgbd)" from="rgb/image"         to="$(arg rgb_topic_relay)"/>
    <remap unless="$(arg subscribe_rgbd)" from="depth/image"       to="$(arg depth_topic_relay)"/>
    <remap unless="$(arg subscribe_rgbd)" from="rgb/camera_info"   to="$(arg camera_info_topic)"/>
    <remap from="rgbd_image"        to="$(arg rgbd_topic_relay)"/>
    <remap from="cloud"             to="voxel_cloud" />

    <param name="decimation"  type="double" value="4"/>
    <param name="voxel_size"  type="double" value="0.0"/>
    <param name="approx_sync" type="bool"   value="$(arg approx_sync)"/>
    <param name="approx_sync_max_interval" type="double" value="$(arg approx_sync_max_interval)"/>
  </node>

</launch>

```

borongyuan, 2023-11-29:
It seems there are some configuration issues. OpenVINS has not been called. Because there are only warning messages from RTAB-Map, but not from OpenVINS.

matlabbe, 2023-12-10:
I'll suggest to launch `stereo_oodmetry` node with `--udebug` argument.
```
<arg name="odom_args"                default="--udebug"/>
```