introlab/rtabmap #1167: Crash when exporting mesh with textures

omgitsraven, 2023-11-25:
I'm having essentially the same issue as #136 , but the issue was old enough that it seemed better to create a new issue...

My steps are:
1. Start capturing (from an Azure Kinect)
2. Stop capturing
3. File > Export 3D Clouds
4. Check "Meshing"
5. Check "Texture Mapping"
6. Set "Texture Format" to ".png"
7. Click "save"
8. The progress bar runs for a while
9. It asks me to save a file, I do
10. It creates a .PNG file containing many images in a grid.  This file looks normal.
11. It creates an .OBJ file
12. Several seconds pass, with the progress window still open
13. RTAB's window (and the progress window) suddenly disappears
14. Trying to open the .OBJ file in blender shows a cloud of vertices, with no faces

matlabbe, 2023-11-26:
Which rtabmap version are you using? Latest release on Windows? Could it be possible to share the database?

omgitsraven, 2023-11-26:
Sorry, yes;  0.21.0 on Windows 10.   Here's a tiny example database just to show that it even happens in such a simple case:

[crashdemo.zip](https://github.com/introlab/rtabmap/files/13467974/crashdemo.zip)

matlabbe, 2023-11-26:
I could not reproduce the issue with this [version](https://github.com/introlab/rtabmap/releases/download/0.21.0/RTABMap-0.21.0-win64.zip) following your instructions. Here is a comparison of the exported OBJ in [CloudCompare](https://www.danielgm.net/cc/) on left, from the point cloud in rtabmap on right: 
[image]

You may try to download the ZIP archive, and start RTABMap.exe from that folder directly, maybe in your current installation, RTABMap.exe is loading a wrong dll on start. You may also enable file logging in Preferences:
[image]
After rtabmap is closed, a file `LogRtabmap.txt` is created in your `~/Documents/RTAB-Map`. 

Note that the black borders in the texture are coming from the RGB image. To crop them we should set "Export 3D clouds dialog" / "Regenerate clouds"->"ROI ratios" to `0.05 0.05 0.05 0.05`: 
[image]

omgitsraven, 2023-11-27:
Downloading a fresh .zip and running it inside there produced the same result.

Here's my log file following the crash:  
[LogRtabmap.txt](https://github.com/introlab/rtabmap/files/13474222/LogRtabmap.txt)

For that matter, here's the .obj and .png that wind up on the disk after the crash:
[crashdemo_export.zip](https://github.com/introlab/rtabmap/files/13474228/crashdemo_export.zip)

matlabbe, 2023-11-28:
Looks like I was testing a version on my build machine using older dependencies. I uploaded [here ](https://github.com/introlab/rtabmap/releases/download/0.21.0/RTABMap-0.21.0-win64-old_deps.zip)to unblock you while I continue investigating what happen with the official release.

omgitsraven, 2023-11-28:
Confirming that your unblocker version successfully works for me too, I've exported a mesh with no crashing now!  Thank you!  (I'll leave the ticket open until it makes it into an official release, I guess?)

matlabbe, 2023-11-28:
Keep it open, I am currently checking again with latest dependencies on windows to see if the issue would be fixed implicitly by a new release.

matlabbe, 2023-11-29:
Confirming with latest dependencies, the next release should not have this issue. Tested with:

[image]

sjmielke, 2024-02-05:
I seem to have the same issue, but I'm on an Apple Silicon Mac. Are there nightly or beta binaries or are .2 or .3 being released soon?

Alternatively, I would optimize, mesh, and export in iOS but even just the scanning can only take small areas before the app crashes, I guess running out of memory. There's no flag or setting to prevent that running out at the expense of processing speed, is there? Thanks!!

matlabbe, 2024-02-17:
I recently added a homebrew formula (https://github.com/Homebrew/homebrew-core/pull/162576) so that it would be easier in the future to build the latest version on mac with:
```
brew install rtabmap --head
```
Nevertheless, I'll try to release new official binaries soon.

sjmielke, 2024-02-18:
Sweet! Can't wait until that's merged, thanks! :)

matlabbe, 2024-02-19:
New binaries here: https://github.com/introlab/rtabmap/releases/tag/0.21.4

closed 2024-02-19 by matlabbe