introlab/rtabmap #1170: Can I load a map and localize the camera and pose within the map?

kenhuang1964, 2023-11-26:
Hey guys, I was wondering if you could help me with something. I'm curious to know if it's possible to load a map, then localize the camera and its pose within that map, and finally, update the map accordingly. Do you think this is something that can be done? Furthermore, is it possible to extract and store the user position path (keyframes)? My ultimate goal is to store the keyframes as nodes (storing coordinates) with edges between them. Also, is april tag available for localization?

matlabbe, 2023-11-26:
Breaking down your message:
 * _"possible to load a map, then localize the camera and its pose within that map"_: yes
 * _"update the map accordingly"_: yes, you would have to manually switch between localization mode and mapping mode after the camera is localized, or start in mapping mode with parameter `Rtabmap/StartNewMapOnLoopClosure` to `true` (this will do the "Append Mode").
 * _"Furthermore, is it possible to extract and store the user position path (keyframes)? My ultimate goal is to store the keyframes as nodes (storing coordinates) with edges between them"_: Do you mean saving a "user graph" during localization mode? No, you would need to save the poses on the side yourself with the current poses computed by rtabmap. 
 * _"is april tag available for localization?"_: yes, either by using external node like apriltag_ros or by enabling `RGBD/MarkerDetection` (if rtabmap is built with opencv's aruco module) and setting `Marker/CornerRefinementMethod=3` and `Marker/Dictionary=20` for 36h11 apriltag family.

cheers,
Mathieu

kenhuang1964, 2023-11-27:
Thank you so much @matlabbe!

closed 2023-11-27 by kenhuang1964

kenhuang1964, 2023-12-09:
Hey @matlabbe, thank you for your help earlier. I have a few more questions if you don't mind:
1) Do you have any recommendations for how to extract the poses computed by rtabmap? Also, what's the difference between optimized poses and poses? 
2) How can I load a map?

Thanks in advance!

matlabbe, 2023-12-10:
Do you use ROS? It will be easier to do what you want with the ROS package.

1. See https://github.com/introlab/rtabmap/issues/1178 for the first question. The optimized poses are saved when rtabmap is closed:
https://github.com/introlab/rtabmap/blob/f56875db4ada45ce9e8564c6a3911ebd809b83bd/corelib/src/Rtabmap.cpp#L518

2. See [localization mode](http://wiki.ros.org/rtabmap_ros/Tutorials/HandHeldMapping#Localization_mode) in ROS. In c++, if you look at the RGBD example, add your database path in `Rtabmap`'s constructor and add parameter `Mem/IncrementalMemory=false` to the `params` map.
https://github.com/introlab/rtabmap/blob/f56875db4ada45ce9e8564c6a3911ebd809b83bd/examples/RGBDMapping/main.cpp#L200-L205