introlab/rtabmap #1174: Recommended Raspberry Pi for running this?

kenhuang1964, 2023-11-30:
Would a 2GB raspberry pi 4B be sufficient or perhaps 4GB? If not, what are some alternatives?

closed 2023-12-04 by kenhuang1964

matlabbe, 2023-12-10:
At least 4 GB, but don't expect high FPS when running the full rtabmap stack on it. If you use only rtabmap node (no VO) at 1 Hz, it can work correctly on RPI4.

miku54, 2024-08-26:
> At least 4 GB, but don't expect high FPS when running the full rtabmap stack on it. If you use only rtabmap node (no VO) at 1 Hz, it can work correctly on RPI4.

How do I set rtabmap node (no VO) at 1 Hz?

matlabbe, 2024-09-05:
Just launch `rtabmap` node, by default it is running at 1Hz already. The "no VO" part is not using `rgbd_odometry` or `stereo_odometry` nodes.