introlab/rtabmap #1175: Using Elevation Mapping instead of OctoMap in ROS-noetic

FlorianMehnert, 2023-12-06:
Hi,

for larger areas it appears that the 3d octomap might become too large to handle. I would like to only use a 2.5d map which should be more efficient in storing the environment data.
Is it possible to substitute the OctoMap used in the rtabmap-slam with the Gridmap used in [Elevation Mapping](https://github.com/ANYbotics/elevation_mapping)?

Thanks :)

matlabbe, 2023-12-10:
You could probably launch `elevation_mapping` node along side of rtabmap node that would provide the TF `map->base_link`, connect for odometry pose to `elevation_mapping` then the robot centric elevation map would be available. If you need the global map as an elevation map and that map is updated on loop closures, we would indeed need to integrate [GridMap](https://github.com/anybotics/grid_map)/[ElevationMap](https://github.com/ANYbotics/elevation_mapping) inside rtabmap like we did with OctoMap. However, the GridMap library seems more focused on local elevation map around the robot (like inside 2-3 meters radius) like a local costmap with rolling window, not a 100 m x 100 m elevation map for example. I don't know if there is maximum map size supported.

I don't have experience with GridMap, so not sure how hard it would be to integrate as an external library. I guess we could assemble the local 3D occupancy grid maps saved in nodes and integrate them as PointCloud2 into a GridMap layer of size equal to our global map. That would be similar than this function:
https://github.com/introlab/rtabmap/blob/f56875db4ada45ce9e8564c6a3911ebd809b83bd/corelib/src/util3d_mapping.cpp#L191
but we would not lose the elevation of the resulting 2D map. And like the [we did](https://github.com/introlab/rtabmap/blob/f56875db4ada45ce9e8564c6a3911ebd809b83bd/corelib/src/OctoMap.cpp#L521) for OctoMap to avoid re-assembling the whole OctoMap when the graph didn't change a lot, the full global map won't need to be regenerated unless bounds are reached or graph optimization makes the nodes move more than X cm.

In term of usage, the GridMap approach could be selected by adding an `updateGridMap` option [here](https://github.com/introlab/rtabmap_ros/blob/0748b91accf9bd173645f321a4ff498caf7ba2b1/rtabmap_util/src/MapsManager.cpp#L456-L457) and updating our GridMap wrapper class. I'll take a look in the code tomorrow.

matlabbe, 2023-12-18:
GridMap has been integrated in PR above for ROS noetic. A new elevation map topic is published. See https://github.com/introlab/rtabmap/pull/1180 for example.

TODO: update ros2 branch with elevation map topic.

FlorianMehnert, 2024-02-07:
Thank you for your effort :heart: 
The integration is working great. I guess I close this issue now

closed 2024-02-07 by FlorianMehnert

matlabbe, 2024-02-07:
Oh yeah the todo for ros2 is indeed done from that merge https://github.com/introlab/rtabmap_ros/commit/80ae448d1b318f5edb6e5cf2360059b468796a6f

Small note, the current implementation will re-update a cell in the elevation map with latest value added to map (there is no average to account the noise, just using latest value). For example, if in one frame there was someone, you may get sudden high elevation, then as soon as the person moves, the values will be reset to floor if in the frame we can see the floor for that particular cell.