introlab/rtabmap #1177: max build error

joara123, 2023-12-08:
Hi I'm getting an error while building on my mac and it's not working
How to remove fatal errors?

[ 56%] Built target rtabmap_utilite
[ 56%] Built target res_tool
[ 56%] Built target rtabmap_core
[ 56%] Building CXX object tools/ConsoleApp/CMakeFiles/consoleApp.dir/main.cpp.o
[ 56%] Building CXX object tools/ExtractObject/CMakeFiles/extractObject.dir/main.cpp.o
[ 56%] Automatic MOC and UIC for target rtabmap_gui
[ 56%] Building CXX object tools/StereoEval/CMakeFiles/stereoEval.dir/main.cpp.o
clang: warning: optimization flag '-ffloat-store' is not supported [-Wignored-optimization-argument]
clang: warning: optimization flag '-ffloat-store' is not supported [-Wignored-optimization-argument]
[ 56%] Building CXX object tools/Camera/CMakeFiles/camera.dir/main.cpp.o
[ 56%] Built target imagesJoiner
clang: warning: optimization flag '-ffloat-store' is not supported [-Wignored-optimization-argument]
clang: warning: optimization flag '-ffloat-store' is not supported [-Wignored-optimization-argument]
[ 56%] Building CXX object tools/KittiDataset/CMakeFiles/kitti_dataset.dir/main.cpp.o
clang: warning: optimization flag '-ffloat-store' is not supported [-Wignored-optimization-argument]
[ 56%] Built target rtabmap_gui_autogen
[ 56%] Building CXX object tools/RgbdDataset/CMakeFiles/rgbd_dataset.dir/main.cpp.o
clang: warning: optimization flag '-ffloat-store' is not supported [-Wignored-optimization-argument]
In file included from /Users/sohui/Desktop/23_2/StudyDesign/SLAM/rtabmap/tools/ExtractObject/main.cpp:28:
In file included from /opt/homebrew/include/pcl-1.13/pcl/point_types.h:354:
In file included from /opt/homebrew/include/pcl-1.13/pcl/impl/point_types.hpp:41:
In file included from /opt/homebrew/include/pcl-1.13/pcl/memory.h:48:
/opt/homebrew/include/eigen3/Eigen/Core:70:10: fatal error: 'omp.h' file not found
#include <omp.h>
         ^~~~~~~
In file included from /Users/sohui/Desktop/23_2/StudyDesign/SLAM/rtabmap/tools/ConsoleApp/main.cpp:30:
In file included from /Users/sohui/Desktop/23_2/StudyDesign/SLAM/rtabmap/corelib/src/../include/rtabmap/core/Rtabmap.h:33:
In file included from /Users/sohui/Desktop/23_2/StudyDesign/SLAM/rtabmap/corelib/src/../include/rtabmap/core/Parameters.h:33:
In file included from /Users/sohui/Desktop/23_2/StudyDesign/SLAM/rtabmap/build/corelib/src/include/rtabmap/core/Version.h:91:
In file included from /opt/homebrew/include/pcl-1.13/pcl/types.h:52:
/opt/homebrew/include/eigen3/Eigen/Core:70:10: fatal error: 'omp.h' file not found
#include <omp.h>
         ^~~~~~~
In file included from /Users/sohui/Desktop/23_2/StudyDesign/SLAM/rtabmap/tools/KittiDataset/main.cpp:28:
In file included from /Users/sohui/Desktop/23_2/StudyDesign/SLAM/rtabmap/corelib/src/../include/rtabmap/core/Odometry.h:33:
In file included from /Users/sohui/Desktop/23_2/StudyDesign/SLAM/rtabmap/corelib/src/../include/rtabmap/core/Transform.h:35:
/opt/homebrew/include/eigen3/Eigen/Core:70:10: fatal error: 'omp.h' file not found
#include <omp.h>
         ^~~~~~~
In file included from /Users/sohui/Desktop/23_2/StudyDesign/SLAM/rtabmap/tools/Camera/main.cpp:28:
In file included from /Users/sohui/Desktop/23_2/StudyDesign/SLAM/rtabmap/corelib/src/../include/rtabmap/core/CameraRGB.h:30:
In file included from /Users/sohui/Desktop/23_2/StudyDesign/SLAM/rtabmap/corelib/src/../include/rtabmap/core/camera/CameraImages.h:30:
In file included from /Users/sohui/Desktop/23_2/StudyDesign/SLAM/rtabmap/corelib/src/../include/rtabmap/core/Camera.h:33:
In file included from /Users/sohui/Desktop/23_2/StudyDesign/SLAM/rtabmap/corelib/src/../include/rtabmap/core/SensorData.h:32:
In file included from /Users/sohui/Desktop/23_2/StudyDesign/SLAM/rtabmap/corelib/src/../include/rtabmap/core/Transform.h:35:
/opt/homebrew/include/eigen3/Eigen/Core:70:10: fatal error: 'omp.h' file not found
#include <omp.h>
         ^~~~~~~
In file included from /Users/sohui/Desktop/23_2/StudyDesign/SLAM/rtabmap/tools/StereoEval/main.cpp:29:
In file included from /Users/sohui/Desktop/23_2/StudyDesign/SLAM/rtabmap/corelib/src/../include/rtabmap/core/Parameters.h:33:
In file included from /Users/sohui/Desktop/23_2/StudyDesign/SLAM/rtabmap/build/corelib/src/include/rtabmap/core/Version.h:91:
In file included from /opt/homebrew/include/pcl-1.13/pcl/types.h:52:
/opt/homebrew/include/eigen3/Eigen/Core:70:10: fatal error: 'omp.h' file not found
#include <omp.h>
         ^~~~~~~
In file included from /Users/sohui/Desktop/23_2/StudyDesign/SLAM/rtabmap/tools/RgbdDataset/main.cpp:28:
In file included from /Users/sohui/Desktop/23_2/StudyDesign/SLAM/rtabmap/corelib/src/../include/rtabmap/core/Odometry.h:33:
In file included from /Users/sohui/Desktop/23_2/StudyDesign/SLAM/rtabmap/corelib/src/../include/rtabmap/core/Transform.h:35:
/opt/homebrew/include/eigen3/Eigen/Core:70:10: fatal error: 'omp.h' file not found
#include <omp.h>
         ^~~~~~~
1 error generated.
make[2]: *** [tools/Camera/CMakeFiles/camera.dir/main.cpp.o] Error 1
make[1]: *** [tools/Camera/CMakeFiles/camera.dir/all] Error 2
make[1]: *** Waiting for unfinished jobs....
1 error generated.
make[2]: *** [tools/StereoEval/CMakeFiles/stereoEval.dir/main.cpp.o] Error 1
make[1]: *** [tools/StereoEval/CMakeFiles/stereoEval.dir/all] Error 2
1 error generated.
make[2]: *** [tools/ConsoleApp/CMakeFiles/consoleApp.dir/main.cpp.o] Error 1
make[1]: *** [tools/ConsoleApp/CMakeFiles/consoleApp.dir/all] Error 2
1 error generated.
make[2]: *** [tools/KittiDataset/CMakeFiles/kitti_dataset.dir/main.cpp.o] Error 1
make[1]: *** [tools/KittiDataset/CMakeFiles/kitti_dataset.dir/all] Error 2
1 error generated.
make[2]: *** [tools/ExtractObject/CMakeFiles/extractObject.dir/main.cpp.o] Error 1
make[1]: *** [tools/ExtractObject/CMakeFiles/extractObject.dir/all] Error 2
1 error generated.
make[2]: *** [tools/RgbdDataset/CMakeFiles/rgbd_dataset.dir/main.cpp.o] Error 1
make[1]: *** [tools/RgbdDataset/CMakeFiles/rgbd_dataset.dir/all] Error 2
make: *** [all] Error 2

matlabbe, 2023-12-10:
Try installing OpenMP: https://formulae.brew.sh/formula/libomp. The error is from third party includes:
```
/opt/homebrew/include/eigen3/Eigen/Core:70:10: fatal error: 'omp.h' file not found
```

joara123, 2023-12-12:
Thank you for your response
I installed OpenMP but the error is still there

joara123, 2023-12-12:
Solved!
I uninstalled and reinstalled and forced the file locations to be moved and it worked

But then I encountered a new error again

[ 56%] Built target rtabmap_utilite
[ 56%] Built target res_tool
[ 56%] Built target rtabmap_core
[ 56%] Automatic MOC and UIC for target rtabmap_gui
[ 56%] Built target imagesJoiner
[ 56%] Built target camera
[ 56%] Built target consoleApp
[ 56%] Built target stereoEval
[ 56%] Built target kitti_dataset
[ 56%] Built target extractObject
[ 56%] Built target rgbd_dataset
[ 56%] Built target recovery
[ 56%] Built target report
[ 56%] Built target detectMoreLoopClosures
[ 56%] Built target export
[ 56%] Built target cleanupLocalGrids
[ 56%] Built target reprocess
[ 56%] Built target info
[ 56%] Built target globalBundleAdjustment
[ 56%] Built target rtabmap_gui_autogen
[ 56%] Automatic RCC for GuiLib.qrc

AutoRcc error
-------------
Locking of the lock file "SRC:/build/guilib/src/CMakeFiles/rtabmap_gui_autogen.dir/AutoRcc_GuiLib_EWIEGA46WW_Lock.lock" failed.
Permission denied

make[2]: *** [guilib/src/rtabmap_gui_autogen/EWIEGA46WW/qrc_GuiLib.cpp] Error 1
make[1]: *** [guilib/src/CMakeFiles/rtabmap_gui.dir/all] Error 2
make: *** [all] Error 2

matlabbe, 2023-12-16:
Are some files have root permissions? Look similar to
https://forum.qt.io/topic/104575/cmake-autogen-of-rcc-list-causes-error-with-permission-denied/3