introlab/rtabmap #1181: Custom VIO into Rtabmap

Genozen, 2023-12-13:


matlabbe, 2023-12-16:
Based on my comment on similar question: https://github.com/introlab/rtabmap_ros/issues/1084#issuecomment-1856752838

What is `/ov_msckf/odomimu`, is it an Odometry msg?
```
ros2 run rtabmap_slam rtabmap --ros-args -r odom:=/ov_msckf/odomimu ...
```

Genozen, 2024-01-16:
Hi, sorry for the duplication on the question (thought my message didn't go through).

And yes, the `/domimu` is a `Odometry msg`

[image]

Would this change anything to your initial suggestion?

I will try it in a couple hours.

Genozen, 2024-01-16:
@matlabbe 

I've remapped some of the topics to match the Openvins one, using the example you gave:

```
ros2 launch rtabmap_launch rtabmap.launch.py \
visual_odometry:=false \
odom_topic:=/ov_msckf/odomimu \
rgb_topic:=/camera/d435i/color/image_raw \
depth_topic:=/camera/d435i/depth/image_rect_raw \
camera_info_topic:=/camera/d435i/color/camera_info \
```

but got a warning message and no image is showing in Rtab:

```
[rtabmap_viz-2] [WARN] [1705445325.096275150] [rtabmap.rtabmap_viz]: rtabmap_viz: Did not receive data since 5 seconds! Make sure the input topics are published ("$ ros2 topic hz my_topic") and the timestamps in their header are set. If topics are coming from different computers, make sure the clocks of the computers are synchronized ("ntpdate"). Parameter "approx_sync" is false, which means that input topics should have all the exact timestamp for the callback to be called.
[rtabmap_viz-2] rtabmap_viz subscribed to (exact sync):
[rtabmap_viz-2]    /ov_msckf/odomimu \
[rtabmap_viz-2]    /camera/d435i/color/image_raw \
[rtabmap_viz-2]    /camera/d435i/depth/image_rect_raw \
[rtabmap_viz-2]    /camera/d435i/color/camera_info
```

I would like to mainly prove out the concept that I can utilize OpenVins' VIO and coupled with Rtabmap to perform loop closure.

I think doing it in the infra1 and infra2 frame might be better? I see there's a D435i launch example that uses the two infra camera img...

Genozen, 2024-01-18:
@borongyuan   Hello, I see you have a PR that was merged regarding OpenVins + Rtabmap. I would love to test this out but am too novice to this. Any chance you could shed some light on how to run your version of the commit? I have a standalone OpenVins and have just properly calibrated my D435i... OV_MSCKF works on a real camera as you did, but am missing the Rtab integration.

borongyuan, 2024-01-18:
If you have an OAK camera, you can try it first. I haven't used RealSense in a long time, so I'm not sure how well it performs. You can first try running RTAB-Map standalone without ROS. The camera driver will parse factory calibration, so you don't need to recalibrate.

matlabbe, 2024-01-20:
@Genozen for this warning:
```
[rtabmap_viz-2] [WARN] [1705445325.096275150] [rtabmap.rtabmap_viz]: rtabmap_viz: Did not receive data since 5 seconds! Make sure the input topics are published ("$ ros2 topic hz my_topic") and the timestamps in their header are set. If topics are coming from different computers, make sure the clocks of the computers are synchronized ("ntpdate"). Parameter "approx_sync" is false, which means that input topics should have all the exact timestamp for the callback to be called.
[rtabmap_viz-2] rtabmap_viz subscribed to (exact sync):
[rtabmap_viz-2]    /ov_msckf/odomimu \
[rtabmap_viz-2]    /camera/d435i/color/image_raw \
[rtabmap_viz-2]    /camera/d435i/depth/image_rect_raw \
[rtabmap_viz-2]    /camera/d435i/color/camera_info
```

Two options:
1. If all those topics are published as expected (verify with `rostopic hz`), you would need to use [rtabmap_sync/rgbd_sync](http://wiki.ros.org/rtabmap_sync#rtabmap_sync.2Frgbd_sync) node to exactly synchronize (with `approx_sync:=false`) your camera topics together, then setup `rtabmap` and `rtabmap_viz` nodes to use `approx_sync:=true` and `subscribe_rgbd:=true`. The nodes will then to try to sync `rgbd_image` topic (output of `rgbd_sync`) with `/ov_msckf/odomimu`.
2. set `odom_frame_id` to frame used by the VIO node as base frame (the frame that would be set as `frame_id` inside `/ov_msckf/odomimu`). The downside is that it will ignore VIO covariance data.

Genozen, 2024-01-24:
Good news, somehow I got the `/rgbd_image` to publish now ~11 to 15 Hz... and no more warning messages. (need to reproduce the steps again)

[image]

The rtabmap_viz seems to update very slow though, any suggestions as to why that is?

matlabbe, 2024-01-27:
What do you mean it is slow? are you seeing only 1 Hz update rate? or is it faster sometimes and lag other time?

You should also try to fix the tf tree, as `canTransform returned after 0.204357 timeout was 0.2.` will always add a 200 ms delay every time the Tf tree is looked at.

Genozen, 2024-02-09:
@matlabbe. Just got back to working on it.

The Rtabmap is indeed reporting 1 Hz. (The synced node, rgbd_image is around 3~5 Hz).

I've added a manual tf topic to link the frames.

code executed:
`ros2 run rtabmap_sync rgbd_sync --ros-args -r /rgb/image:=/camera/d435i/color/image_raw -r /depth/image:=/camera/d435i/depth/image_rect_raw -r /rgb/camera_info:=/camera/d435i/color/camera_info -p approx_sync:=false`

`ros2 run rtabmap_slam rtabmap --ros-args -r odom:=/ov_msckf/odomimu -r /depth/image:=/camera/d435i/depth/image_rect_raw -r /rgb/image:=/camera/d435i/color/image_raw -r /rgb/camera_info:=/camera/d435i/camera_info -p approx_sync:=true -p subscribe_rgbd:=true -p frame_id:="d435i_link"`

There's also a g2o related issue that might not be related to contributing the slowness of rtabmap
```
[INFO] [1707505685.195324591] [rtabmap]: rtabmap(maps): latch                      = true
[INFO] [1707505685.195665228] [rtabmap]: rtabmap(maps): map_filter_radius          = 0.000000
[INFO] [1707505685.195699111] [rtabmap]: rtabmap(maps): map_filter_angle           = 30.000000
[INFO] [1707505685.195716058] [rtabmap]: rtabmap(maps): map_cleanup                = true
[INFO] [1707505685.195731333] [rtabmap]: rtabmap(maps): map_always_update          = false
[INFO] [1707505685.195743214] [rtabmap]: rtabmap(maps): map_empty_ray_tracing      = true
[INFO] [1707505685.195754468] [rtabmap]: rtabmap(maps): cloud_output_voxelized     = true
[INFO] [1707505685.195765399] [rtabmap]: rtabmap(maps): cloud_subtract_filtering   = false
[INFO] [1707505685.195776294] [rtabmap]: rtabmap(maps): cloud_subtract_filtering_min_neighbors = 2
[INFO] [1707505685.195854403] [rtabmap]: rtabmap(maps): octomap_tree_depth         = 16
[INFO] [1707505685.221103171] [rtabmap]: rtabmap: frame_id      = d435i_link
[INFO] [1707505685.221153846] [rtabmap]: rtabmap: map_frame_id  = map
[INFO] [1707505685.221164050] [rtabmap]: rtabmap: log_to_rosout_level  = 4
[INFO] [1707505685.221172713] [rtabmap]: rtabmap: initial_pose  = 
[INFO] [1707505685.221182062] [rtabmap]: rtabmap: use_action_for_goal  = false
[INFO] [1707505685.221190412] [rtabmap]: rtabmap: tf_delay      = 0.050000
[INFO] [1707505685.221202767] [rtabmap]: rtabmap: tf_tolerance  = 0.100000
[INFO] [1707505685.221210896] [rtabmap]: rtabmap: odom_sensor_sync   = false
[INFO] [1707505685.221218388] [rtabmap]: rtabmap: pub_loc_pose_only_when_localizing = false
[INFO] [1707505685.221226271] [rtabmap]: rtabmap: gen_scan  = false
[INFO] [1707505685.221233731] [rtabmap]: rtabmap: gen_depth  = false
[INFO] [1707505685.353250273] [rtabmap]: RTAB-Map detection rate = 1.000000 Hz
[INFO] [1707505685.353386668] [rtabmap]: rtabmap: Using database from "/home/kishore/kishorenukala/.ros/rtabmap.db" (0 MB).
[INFO] [1707505685.372638490] [rtabmap]: rtabmap: Database version = "0.21.4".
[INFO] [1707505685.372815588] [rtabmap]: rtabmap: SLAM mode (Mem/IncrementalMemory=true)
[INFO] [1707505685.399605817] [rtabmap]: Setup callbacks
[WARN] [1707505685.399715832] [rtabmap]: rtabmap: Parameters subscribe_depth and subscribe_rgbd cannot be true at the same time. Parameter subscribe_depth is set to false.
[INFO] [1707505685.399824459] [rtabmap]: rtabmap: subscribe_depth = false
[INFO] [1707505685.399849595] [rtabmap]: rtabmap: subscribe_rgb = false
[INFO] [1707505685.399866216] [rtabmap]: rtabmap: subscribe_stereo = false
[INFO] [1707505685.399880158] [rtabmap]: rtabmap: subscribe_rgbd = true (rgbd_cameras=1)
[INFO] [1707505685.399895624] [rtabmap]: rtabmap: subscribe_sensor_data = false
[INFO] [1707505685.399909033] [rtabmap]: rtabmap: subscribe_odom_info = false
[INFO] [1707505685.399922833] [rtabmap]: rtabmap: subscribe_user_data = false
[INFO] [1707505685.399935344] [rtabmap]: rtabmap: subscribe_scan = false
[INFO] [1707505685.399947196] [rtabmap]: rtabmap: subscribe_scan_cloud = false
[INFO] [1707505685.399959215] [rtabmap]: rtabmap: subscribe_scan_descriptor = false
[INFO] [1707505685.399971328] [rtabmap]: rtabmap: queue_size      = 10
[INFO] [1707505685.399983224] [rtabmap]: rtabmap: qos_image       = 0
[INFO] [1707505685.399995144] [rtabmap]: rtabmap: qos_camera_info = 0
[INFO] [1707505685.400006427] [rtabmap]: rtabmap: qos_scan        = 0
[INFO] [1707505685.400017970] [rtabmap]: rtabmap: qos_odom        = 0
[INFO] [1707505685.400029566] [rtabmap]: rtabmap: qos_user_data   = 0
[INFO] [1707505685.400040777] [rtabmap]: rtabmap: approx_sync     = true
[INFO] [1707505685.400065318] [rtabmap]: Setup rgbd callback
[INFO] [1707505685.401936115] [rtabmap]: 
rtabmap subscribed to (approx sync):
   /ov_msckf/odomimu \
   /rgbd_image
[INFO] [1707505685.562125497] [rtabmap]: rtabmap (1): Rate=1.00s, Limit=0.000s, Conversion=0.0094s, RTAB-Map=0.0432s, Maps update=0.0001s pub=0.0000s (local map=1, WM=1)
[INFO] [1707505686.554623626] [rtabmap]: rtabmap (2): Rate=1.00s, Limit=0.000s, Conversion=0.0008s, RTAB-Map=0.0464s, Maps update=0.0000s pub=0.0000s (local map=1, WM=1)
[INFO] [1707505687.592417314] [rtabmap]: rtabmap (3): Rate=1.00s, Limit=0.000s, Conversion=0.0005s, RTAB-Map=0.0476s, Maps update=0.0000s pub=0.0000s (local map=1, WM=1)
[INFO] [1707505688.592018329] [rtabmap]: rtabmap (4): Rate=1.00s, Limit=0.000s, Conversion=0.0010s, RTAB-Map=0.0476s, Maps update=0.0001s pub=0.0000s (local map=1, WM=1)
[INFO] [1707505689.584141384] [rtabmap]: rtabmap (5): Rate=1.00s, Limit=0.000s, Conversion=0.0013s, RTAB-Map=0.0381s, Maps update=0.0000s pub=0.0000s (local map=1, WM=1)
[... cut ...]
[FATAL] (2024-02-09 13:08:47.964) OptimizerG2O.cpp:1835::optimizeBA() Condition (optimizer.verifyInformationMatrices()) not met!
[FATAL] [1707505727.964122881] [rtabmap]: [FATAL] (2024-02-09 13:08:47.964) OptimizerG2O.cpp:1835::optimizeBA() Condition (optimizer.verifyInformationMatrices()) not met!
terminate called after throwing an instance of 'UException'
  what():  [FATAL] (2024-02-09 13:08:47.964) OptimizerG2O.cpp:1835::optimizeBA() Condition (optimizer.verifyInformationMatrices()) not met!
[ros2run]: Aborted
```

KishoreNukala, 2024-02-20:
rgbd_image is generated using rtabmap_sync node as explained in previous message and /ov_msckf/odomimu is generated from openvins. The rtabmapslam node ends up with the graph optimisation error all the time. Whenever the rtabmap slam hits the optimizer error, the rviz visualisation of openvins freezes for 2 sec and odomimu path drifts later  in rviz visualisation.
The commands used are:
`ros2 launch ov_msckf subscribe.launch.py config:=rs_d455`
`ros2 run rtabmap_sync rgbd_sync --ros-args -r /rgb/image:=/camera/d435i/color/image_raw -r /depth/image:=/camera/d435i/depth/image_rect_raw -r /rgb/camera_info:=/camera/d435i/color/camera_info -p approx_sync:=false -p queue_size:=10`
`ros2 run rtabmap_slam rtabmap --ros-args    -r odom:=/ov_msckf/odomimu    -p approx_sync:=true    -p subscribe_rgbd:=true    -p frame_id:="d435i_link"`

ros bag with `/tf /tf_static  /ov_msckf/odomimu /rgbd_image`
[https://drive.google.com/file/d/1QYWSC7kC7-mgr8KF2XhGcv5JKjkBZIW-/view?usp=sharing](url)

matlabbe, 2024-02-24:
The odometry is not computed in the right frame. We can see the resulting point clouds have an optical rotation in them (the floor is vertical instead of being flat):
[image]

I guess `global`, `imu`, `cam0` and `cam1` frames are published by `ov_msckf` node? Is `ov_msckf` ignoring the frame set in image topic or `d435i_link`?

I'll suggest to debug the cloud orientation with rviz first by launching only `ov_msckf` and visualize the live camera point cloud. Debug `ov_msckf` till the point cloud is correctly oriented, then retry rtabmap.

Kishore-1297, 2024-02-25:
Hi @matlabbe Thank you for your reply. I have checked the point cloud being generated by d435i and observed that it is being generated in `d435i_depth_optical_frame` and also a static tf is always being published between frames `d435i_link` and `d435i_depth_optical_frame`. When point cloud is visualised **without** the ov_msckf running in the background the live point cloud looks correct wrt all the axes and `d435i_link` being the reference frame

When `ov_msckf` is executed, `global, imu, cam0, and cam1` frames are generated. And the tree is incomplete as there is no `tf` between these frames and `d435i...` frames. To avoid the warnings in `tf` I have run a static tf publisher `ros2 run tf2_ros static_transform_publisher 0 0 0 0 0 0 "imu" "d435i_link"`. The pointcloud is still being generated in `d435i_depth_optical_frame` as in the previous case but when visualised in rviz the point cloud is optically rotated in global frame due to the `tf` that is already in place between `imu` and `global`. This same optically rotated point cloud  is being reflected in rtabmap-rviz eventhough the `frame_id` for rtab was set to `d435i_link`.  

I could get the live point cloud visualisation correctly by defining a static identity tf `imu` and `d435i_depth_optical_frame` instead of `d435i_link`,  but I know that this isn't a correct way of doing it as it would lead to tf errors between `global` and `d435i_color_optical_frame` while running RTAB MAP .And `/ov_msckf/odomimu` is being generated in `global` frame.

I don't have much of an experience working and debugging on tf and would greatly appreciate your inputs regd it.

matlabbe, 2024-02-26:
> I could get the live point cloud visualisation correctly by defining a static identity tf imu and d435i_depth_optical_frame instead of d435i_link

You can then set `imu` as the base frame `frame_id` in rtabmap, that could work.