introlab/rtabmap #1183: Ignore landmarks for optimisation

hellovuong, 2023-12-15:
Hello,
I am starting to use Apriltag bundles as landmarks while mapping. After saved the map and use it for localization. Most of the time it works fine to recognise and correct the position. However, at some point, rtabmap starts to reject the landmark link according to this log:
```
[ WARN] (2023-12-11 21:50:52.944) Rtabmap.cpp:3193::process() Rejecting localization (11899 <-> -402) in this iteration because a wrong loop closure has been detected after graph optimization, resulting in a maximum graph error ratio of 227.124359 (edge 11896->11897, type=0, abs error=7.182302 m, stddev=0.031623). The maximum error ratio parameter "RGBD/OptimizeMaxError" is 3.000000 of std deviation.
[ WARN] (2023-12-11 21:50:52.944) Rtabmap.cpp:3222::process() Rejecting localization (11899 <-> -402) in this iteration because a wrong loop closure has been detected after graph optimization, resulting in a maximum graph error ratio of 21.780865 (edge 11894->11895, type=0, abs error=39.463692 deg, stddev=0.031623). The maximum error ratio parameter "RGBD/OptimizeMaxError" is 3.000000 of std deviation.
[ WARN] (2023-12-11 21:50:53.576) Rtabmap.cpp:3193::process() Rejecting localization (11900 <-> -402) in this iteration because a wrong loop closure has been detected after graph optimization, resulting in a maximum graph error ratio of 221.190506 (edge 11898->11899, type=0, abs error=6.994658 m, stddev=0.031623). The maximum error ratio parameter "RGBD/OptimizeMaxError" is 3.000000 of std deviation.
[ WARN] (2023-12-11 21:50:53.576) Rtabmap.cpp:3222::process() Rejecting localization (11900 <-> -402) in this iteration because a wrong loop closure has been detected after graph optimization, resulting in a maximum graph error ratio of 18.399179 (edge 11899->11900, type=0, abs error=33.336583 deg, stddev=0.031623). The maximum error ratio parameter "RGBD/OptimizeMaxError" is 3.000000 of std deviation.
```
 I came across the parameter:
`Optimizer/LandmarksIgnore 
` when I set it to true on the database that create with its value = false (if it is relevant) g2o crashed immediately with invalid Hessian block.
I am wondering if we can return absolute position of robot and its covariance if it detects landmarks?
Thank you!

matlabbe, 2023-12-16:
It seems the tag detection is deforming too much the graph based on odometry covariance (0.031623m error). You can set `RGBD/OptimizeMaxError` to 0 to see if the tag detection does indeed deform the graph. If it does, then tag detection may be not super accurate. Note that apriltag detection orientation estimation is poor the farther from the tag you are, you could filter by distance the tag detections.

hellovuong, 2024-02-13:
Thank you for your reply.

closed 2024-02-13 by hellovuong