introlab/rtabmap #1189: performance difference when running superglue from python and rtabmap_matcher

agodlin, 2023-12-26:
Hello
i have noticed X2 slow down on CPU when exectuting results = superglue(data) in rtabmap_superglue.py

i am passing 2k keypoints for the From and To data.

when running python3 rtabmap_superglue.py this takes 13.7s on avg
but when i run rtabmap-matcher --Vis/CorNNType 6 --Vis/FeaturesType 1 --PyMacher/Path /work/rtabmap/corelib/src/python/rtabmap_superglue.py img1.png img1.png
i get 25.5s run time for same line of code. 

i did see same behivour with a different CNN model as to make sure it isn't model specific problem. 

can't understand why there is this slow down, i hard coded in rtabmap_superglue.py to call for match several times inside at the end of the init() function just to make sure it isn't some cpp-python overhead slowdown.

matlabbe, 2024-01-02:
We got a related comment: https://github.com/introlab/rtabmap/issues/1148#issuecomment-1860792470 about performance issue with python code from c++. No fix or workaround found yet.

agodlin, 2024-01-03:
> We got a related comment: [#1148 (comment)](https://github.com/introlab/rtabmap/issues/1148#issuecomment-1860792470) about performance issue with python code called from c++. No fix or workaround found yet.
[CMakeLists.txt](https://github.com/introlab/rtabmap/files/13817623/CMakeLists.txt)
[main.txt](https://github.com/introlab/rtabmap/files/13817624/main.txt)

Hi, thanks for the respond. that issue really look similar

but i am not sure the problem is in "performance issue with python code called from c++"
i made simple c++ test that is using "same" code (copy paste code + hard coded some stuff) as the one in PyMatcher.cpp 
and i get same performance of 13.7s just as when i use python script directly so it isnt c++ calling python problem as i understand it

i suspect this could be some cmake configuration that have effect on this, or maybe some other pkg that is used in rtabmap. 

adding the cmake + main file of the test if you want to look, i changed also rtabmap_superglue.py to run the matches on init with some hard coded keypoints and random descriptors 

if you have any ideas what i could check i would happy to test and report back.