introlab/rtabmap #1194: rtabmap+vinsfusion,The problem encountered when running rtabmap.launch.

t1tqq, 2024-01-03:
hi!
  I am trying to use my Realsense camera with rtabmap, and I ran the following command: 
       roslaunch rtabmap_launch rtabmap.launch rtabmap_args:="--delete_db_on_start" left_image_topic:=/stereo/left/image_raw right_image_topic:=/stereo/right/image_raw left_camera_info_topic:=/stereo/left/camera_info right_camera_info_topic:=/stereo/right/camera_info stereo:=true wait_imu_to_init:=true imu_topic:=/rtabmap/imu

  I received some errors and warnings, but my rtabmap_viz seems to be running. 
   [ WARN] [1704283006.882115782]: Could not get transform from world to camera_link after 0.200000 seconds (for stamp=1704283005.673620)! Error="Could not find a connection between 'world' and 'camera_link' because they are not part of the same tree.Tf has two or more unconnected trees.. canTransform returned after 0.200057 timeout was 0.2.".
  [ WARN] [1704283006.903989118]: Could not get transform from world to camera_infra1_optical_frame after 0.200000 seconds (for stamp=1704283006.339447)! Error="Could not find a connection between 'world' and 'camera_infra1_optical_frame' because they are not part of the same tree.Tf has two or more unconnected trees.. canTransform returned after 0.201672 timeout was 0.2.".
 [ERROR] [1704283006.904053123]: TF of received image 0 at time 1704283006.339447s is not set!
 [ERROR] [1704283006.904104808]: Could not convert rgb/depth msgs! Aborting rtabmap_viz update...
[image]

Based on the warnings, it seems that there is a missing transformation from the world frame to another coordinate system, but I'm not sure how to fix it. Here are my TF tree and graph. 
[image]
[image]

Can you tell what might be causing this issue? If you need any additional data, I can provide it. I hope you can help me, thank you!

matlabbe, 2024-01-03:
The vins estimator should use `camera_link` as camera frame instead of `camera`. Maybe there is a config in vins to change it.

t1tqq, 2024-02-25:
Thank you for your reply. I have already solved that issue. 
However, it seems that the map I obtained has some problems. 
This map is saved in ~/.ros/rtabmap.db. When I check Cloud and Odom under the image, the map coordinate system appears to be incorrect.
[image]

 But when I only check Cloud, the map looks fine. 
[image]

Is this a normal phenomenon, or do I need to perform coordinate transformations somewhere? Another question is, I plan to use this map for navigation in the next step. How can I utilize rtabmap.db? What should I do next? Could you please give me some help or suggestions? Thank you!

matlabbe, 2024-02-26:
That `Odom` checkbox is to see the cloud in `odom` frame (roll, pitch) instead of base frame. That cloud is only visualizing a single node in the map. Open the database in rtabmap app instead, it will show the 3d point cloud of all nodes in map frame.

To use the map for navigation, we often relaunch rtabmap in localization mode. See Advanced section [here](http://wiki.ros.org/rtabmap_ros/Tutorials/HandHeldMapping#Localization_mode).

t1tqq, 2024-02-26:
I have already saved rtabmap.db. Could you please provide more details on how to open the database in the rtabmap app? What is this rtabmap app?

matlabbe, 2024-02-27:
Are you using ROS?
```
$ rtabmap rtabmap.db
```

t1tqq, 2024-02-27:
"Yes, I am using ROS. I tried the command $ rtabmap ~/.ros/rtabmap.db, but encountered a series of errors."

[ WARN] (2024-02-27 15:41:28.648) MainWindow.cpp:5562::openDatabase() Parameter "RGBD/CreateOccupancyGrid": Database="true" Preferences="false"
[ WARN] (2024-02-27 15:41:30.906) StatsToolBox.cpp:649::setWorkingDirectory() The directory "/home/lw/Documents/RTAB-Map" doesn't exist, using "/home/lw" instead...
[ERROR] (2024-02-27 15:41:30.907) Rtabmap.cpp:4679::setWorkingDirectory() Directory "/home/lw/Documents/RTAB-Map" doesn't exist!
[ERROR] (2024-02-27 15:41:32.526) Rtabmap.cpp:4679::setWorkingDirectory() Directory "/home/lw/Documents/RTAB-Map" doesn't exist!
[ERROR] (2024-02-27 15:42:17.324) CameraOpenNI2.cpp:211::init() CameraOpenNI2: No device detected!

matlabbe, 2024-03-02:
The odometry pose looks wrong:
[image]

For that that particular node by looking at the actual image, I would have expected a roll and pitch value around 0.1 rad at most (slightly rotated on right, slight pitch towards ground), not -1.88 rad in roll.

It feels like the odometry is giving the pose of the IMU frame instead of the camera frame. There could be a 90 deg offset between these two frames. You could change the `frame_id` of rtabmap node to use the same frame than vins-fusion is using (e.g., `imu` frame).

lifafang, 2025-10-29:
Could you share the process of configuring VINS Fusion for Rtabmap? @t1tqq