introlab/rtabmap #1198: Map correction decays to identity when static

DennisMelamedKitware, 2024-01-15:
Hi!

I'm currently using RTABMAP as a c++ library in an application I'm building, following a very similar structure as your C++ RGBD mapping example. I am using an aruco tag as the origin of my map coordinate frame, and tracking the location of a realsense D455. When my application starts running and first detects the aruco, I get a very reasonable map correction from `odometryCorrection_ = stats.mapCorrection();`, where stats is the `rtabmap::Statistics` object that the `rtabmapThread` emits. However, if I let the camera sit without moving, over time the `odometryCorrection` appears to decay to identity. The odometry predicted pose itself stays consistently near identity (as expected). 

Is the correction decaying to identity expected behavior? I feel I am misunderstanding something about the structure of rtabmap and the optimizations being run. I've attached my config.ini (renamed to config.txt) file that I am loading parameters from here [config.txt](https://github.com/introlab/rtabmap/files/13942974/config.txt).

Thanks!
Dennis

matlabbe, 2024-01-20:
the odometry correction would not decay to identity, unless odometry was indeed perfect, but the tag detection was wrong... and the more tag detections you have, the more you can correct the tag orientation (which individually are often bad), thus slowly "fix" the initial bad tag orientation error.

DennisMelamedKitware, 2024-02-12:
I think I'm starting to understand the issue here - odometry is assuming it starts at the origin (identity transform), which doesn't match the fact that I've set one of my arucos to be the origin (set the prior pose to 0,0,0,0,0,0). Eventually this could cause the decay I mention above, I think. 

Is there any ability in rtabmap standalone to set the initial pose of the camera/robot in mapping mode? If I set it to be the correct transform relative to the tag I'm using as my map origin, that should lead to a consistent estimation (assuming everything else is correct), right?  From this (https://github.com/introlab/rtabmap/blob/a11ea291d8391676cf2ddc83f110c89fc237b9e4/corelib/src/Rtabmap.cpp#L852) it appears we can only set the initial pose in localization mode, is that accurate?

matlabbe, 2024-02-17:
Odometry will always start at (0,0) in its odometry frame (i.e.: "odom -> base_link" would be Identity on startup). On the mapping side, the "map -> odom" transform will be Identity till a loop closure is detected.

`Marker/Priors` parameter can be used to set a "world" coordinate to a tag. To include that prior when detecting a tag, we should set `Optimizer/PriorsIgnored=false`. So for:

> The map will be transformed in marker's world frame when a tag is detected. 

it means that the initial pose of the camera will be Identity until a tag is detected. When the tag is detected, the graph will be transformed in tag's coordinate frame, so that "map->odom" will be updated to make the camera pose (map -> odom -> base_link) appears in tag's frame. For example: If you bringup the camera, move forward 1 meter along x axis (so that odom->base_link is now x=1, y=0), then detect a tag 1 meter away with a prior at (x=10 y=0), the "map -> odom" will be updated to( x=8, y=0), so that the camera pose in the map frame "map->odom->base_link" will be (x=9, y=0).

DennisMelamedKitware, 2024-02-20:
@matlabbe thanks for the clarification - the "map->odom->base_link" transform is what is computed in this example (https://github.com/introlab/rtabmap/wiki/Cplusplus-RGBD-Mapping) by `odometryCorrection_*odom.pose()`, right?

matlabbe, 2024-02-24:
Yes, `map->odom` would be `odometryCorrection_`, and `odom->base_link` would be `odom.pose()`.

closed 2024-02-26 by DennisMelamedKitware