introlab/rtabmap #1199: How to get apriltags' corner location and ID

ZitongLan, 2024-01-16:
Hello! I am new to rtabmap. And I put some tags in the environment to enhance the map building quality. But actually what I am interesting know is the tag's location in the world coordinate. I turn on the tag detection option in the rtabmap and have successfully recognized the tags. I saved the map data into a .db file. Is there a way to get the tags information from these .db file? **For example, the tags' ID and location of the tags' four corners in the world coordinates?** 

[image]

ZitongLan, 2024-01-16:
[image]
for example, I have the above tag detection. And I would like to know the tags' id and their corredponding transformation matrix as showed in the map.

ZitongLan, 2024-01-16:
https://drive.google.com/file/d/1bjzE68mvXHUCNhVxd3hRkiu6eC-2NTSz/view?usp=drive_link
to help me understand my .db file. I attached the original tag_detect.db file here

matlabbe, 2024-01-20:
To make it more convenient, I updated the menu option "File->Export poses..." with commit above to easily export tag poses with RGBDSLAM +ID format and g2o format. 

g2o format:
```
PARAMS_SE3OFFSET 0 0.000000 0.000000 0.000000 0.000000 0.000000 0.000000 1.000000
VERTEX_SE3:QUAT 42 0.819282 0.488257 0.011752 -0.441424 -0.496760 0.563409 0.490860
VERTEX_SE3:QUAT 38 0.522782 0.837418 0.015320 -0.199097 -0.630587 0.713700 0.230983
VERTEX_SE3:QUAT 37 0.904054 -0.151404 0.013637 -0.481060 -0.400194 0.503236 0.595970
VERTEX_SE3:QUAT 35 0.832135 -0.533397 0.018474 -0.605095 -0.301754 0.334092 0.656648
VERTEX_SE3:QUAT 1 -0.000000 -0.000000 -0.000000 -0.011382 -0.040387 0.000323 0.999119
VERTEX_SE3:QUAT 2 0.000577 -0.000267 -0.000228 -0.011366 -0.040431 0.000465 0.999118
VERTEX_SE3:QUAT 3 -0.000539 -0.000272 0.000131 -0.011343 -0.040275 0.000326 0.999124
VERTEX_SE3:QUAT 4 -0.000850 -0.000450 0.000132 -0.011571 -0.040339 0.000345 0.999119
VERTEX_SE3:QUAT 5 -0.000607 -0.000695 0.000101 -0.011400 -0.040553 0.000428 0.999112
VERTEX_SE3:QUAT 6 -0.000809 -0.000266 -0.000064 -0.011428 -0.040695 0.000220 0.999106
VERTEX_SE3:QUAT 7 -0.000564 -0.000646 0.000295 -0.011289 -0.040402 0.000537 0.999120
VERTEX_SE3:QUAT 8 -0.000437 -0.000777 0.000012 -0.011142 -0.040316 0.000868 0.999125
VERTEX_SE3:QUAT 9 -0.000489 -0.001396 0.000457 -0.010361 -0.039980 0.001563 0.999146
VERTEX_SE3:QUAT 10 -0.000609 -0.000616 0.000154 -0.009053 -0.040920 0.010313 0.999068
VERTEX_SE3:QUAT 11 -0.006241 0.011677 0.000039 -0.004871 -0.040644 0.110201 0.993066
VERTEX_SE3:QUAT 12 -0.006476 0.015727 0.000844 -0.011053 -0.036330 0.114536 0.992693
VERTEX_SE3:QUAT 13 -0.011113 0.013309 0.001670 0.003570 -0.036042 0.117771 0.992380
VERTEX_SE3:QUAT 14 -0.016918 0.017398 0.000595 0.006427 -0.037218 0.193481 0.980377
VERTEX_SE3:QUAT 15 -0.018686 0.020766 0.000219 0.001921 -0.038344 0.205386 0.977928
VERTEX_SE3:QUAT 16 -0.017305 0.017686 0.000023 0.010132 -0.037168 0.198688 0.979305
VERTEX_SE3:QUAT 17 -0.023873 0.006940 0.009873 0.014546 -0.021110 0.107169 0.993910
VERTEX_SE3:QUAT 18 -0.025634 0.001384 0.007292 -0.025846 -0.036911 -0.027053 0.998618
VERTEX_SE3:QUAT 19 -0.026471 -0.003155 0.001741 -0.015777 -0.041427 -0.078213 0.995951
VERTEX_SE3:QUAT 20 -0.027618 -0.006194 0.002638 -0.017402 -0.044063 -0.144016 0.988441
VERTEX_SE3:QUAT 24 -0.034211 -0.001495 0.002570 0.010180 -0.035211 0.030620 0.998859
VERTEX_SE3:QUAT 25 -0.034994 0.021265 0.002141 -0.022261 -0.022208 0.163951 0.985967
EDGE_SE3 42 1 -0.383925 0.107978 0.872058 0.415670 0.485312 -0.570876 0.515550 999.999953 0.000000 0.000000 0.000000 0.000000 0.000000 999.999953 0.000000 0.000000 0.000000 0.000000 999.999953 0.000000 0.000000 0.000000 100.000002 0.000000 0.000000 100.000002 0.000000 100.000002
EDGE_SE3 42 2 -0.383925 0.107978 0.872058 0.415670 0.485312 -0.570876 0.515550 999.999953 0.000000 0.000000 0.000000 0.000000 0.000000 999.999953 0.000000 0.000000 0.000000 0.000000 999.999953 0.000000 0.000000 0.000000 100.000002 0.000000 0.000000 100.000002 0.000000 100.000002
EDGE_SE3 42 3 -0.383925 0.107978 0.872058 0.415670 0.485312 -0.570876 0.515550 999.999953 0.000000 0.000000 0.000000 0.000000 0.000000 999.999953 0.000000 0.000000 0.000000 0.000000 999.999953 0.000000 0.000000 0.000000 100.000002 0.000000 0.000000 100.000002 0.000000 100.000002
EDGE_SE3 42 4 -0.369929 0.107138 0.878564 0.412710 0.489235 -0.574260 0.510437 999.999953 0.000000 0.000000 0.000000 0.000000 0.000000 999.999953 0.000000 0.000000 0.000000 0.000000 999.999953 0.000000 0.000000 0.000000 100.000002 0.000000 0.000000 100.000002 0.000000 100.000002
EDGE_SE3 42 5 -0.376693 0.124524 0.870749 0.409218 0.481601 -0.577319 0.517014 999.999953 0.000000 0.000000 0.000000 0.000000 0.000000 999.999953 0.000000 0.000000 0.000000 0.000000 999.999953 0.000000 0.000000 0.000000 100.000002 0.000000 0.000000 100.000002 0.000000 100.000002
EDGE_SE3 42 6 -0.376693 0.124524 0.870749 0.409218 0.481601 -0.577319 0.517014 999.999953 0.000000 0.000000 0.000000 0.000000 0.000000 999.999953 0.000000 0.000000 0.000000 0.000000 999.999953 0.000000 0.000000 0.000000 100.000002 0.000000 0.000000 100.000002 0.000000 100.000002
EDGE_SE3 42 7 -0.376693 0.124524 0.870749 0.409218 0.481601 -0.577319 0.517014 999.999953 0.000000 0.000000 0.000000 0.000000 0.000000 999.999953 0.000000 0.000000 0.000000 0.000000 999.999953 0.000000 0.000000 0.000000 100.000002 0.000000 0.000000 100.000002 0.000000 100.000002
EDGE_SE3 42 8 -0.376693 0.124524 0.870749 0.409218 0.481601 -0.577319 0.517014 999.999953 0.000000 0.000000 0.000000 0.000000 0.000000 999.999953 0.000000 0.000000 0.000000 0.000000 999.999953 0.000000 0.000000 0.000000 100.000002 0.000000 0.000000 100.000002 0.000000 100.000002
EDGE_SE3 42 9 -0.403238 0.101630 0.857586 0.421404 0.481304 -0.563134 0.523111 999.999953 0.000000 0.000000 0.000000 0.000000 0.000000 999.999953 0.000000 0.000000 0.000000 0.000000 999.999953 0.000000 0.000000 0.000000 100.000002 0.000000 0.000000 100.000002 0.000000 100.000002
EDGE_SE3 42 10 -0.353608 0.115757 0.889999 0.410994 0.485816 -0.577996 0.510869 999.999953 0.000000 0.000000 0.000000 0.000000 0.000000 999.999953 0.000000 0.000000 0.000000 0.000000 999.999953 0.000000 0.000000 0.000000 100.000002 0.000000 0.000000 100.000002 0.000000 100.000002
EDGE_SE3 42 11 -0.400221 0.110002 0.849665 0.479976 0.418094 -0.508716 0.579679 999.999953 0.000000 0.000000 0.000000 0.000000 0.000000 999.999953 0.000000 0.000000 0.000000 0.000000 999.999953 0.000000 0.000000 0.000000 100 [... cut ...]
1705439464.111071 -0.025634 0.001384 0.007292 -0.025846 -0.036911 -0.027053 0.998618 18
1705439465.140646 -0.026471 -0.003155 0.001741 -0.015777 -0.041427 -0.078213 0.995951 19
1705439466.170138 -0.027618 -0.006194 0.002638 -0.017402 -0.044063 -0.144016 0.988441 20
1705439470.251940 -0.034211 -0.001495 0.002570 0.010180 -0.035211 0.030620 0.998859 24
1705439471.255901 -0.034994 0.021265 0.002141 -0.022261 -0.022208 0.163951 0.985967 25
```
Note that to keep this format compatible with some benchmark tools, I put al landmarks with stamp=0 to be easily ignored downstream if needed.

closed 2025-04-01 by ZitongLan