introlab/rtabmap #1204: creating a .db file from .pcd map

jhb-ipu, 2024-01-25:
Hi,
I am trying to use Rtabmap in localization mode. The scenario is that I have a high resolution 3D coloured pointcloud as a map that was created with a different scanner than the one on my robot. It is not possible to feed the data from the high resolution scanner directly into Rtabmap or ROS. The scanner can produce .pcd or .ply format.

I saw that it is possible to export .db files as .pcd but my scenario requires the reverse of this. Is there any way to do this with Rtabmap or do you have tips how I could convert the data from .pcd to .db myself?

matlabbe, 2024-01-27:
That is not possible.

A hacky way would be to create a mapping session from your single PCD file, so that one node will be created in the map. Localization would be done always against that same node all the time. That is however not trivial to do and not super convenient to use as the initial localization would need to manually be set, and ICP parameters carefully tuned as you may not want to do ICP on a high resolution 3D point cloud.

jhb-ipu, 2024-01-29:
Thank you for the tips on how to deal with this problem and the great work on Rtabmap.

closed 2024-01-29 by jhb-ipu