introlab/rtabmap #1205: Compression Algorithm for UserData

360wcui, 2024-01-25:
Hello,  I am trying to read user_data from rtabmap.db directly, and notice that the data is compressed:  https://github.com/introlab/rtabmap/blob/b374c6cd8e9d1046277555c58ff0fe1b81e3bff1/corelib/src/SensorData.cpp#L422

Which compression algorithm do you use,  what's the rationale behind using the compression algorithm for UserData.  Would it be possible to disable compression?  Thank you.

360wcui, 2024-01-25:
The compression uses zlib library, which employs the DEFLATE compression algorithm.

closed 2024-01-25 by 360wcui

matlabbe, 2024-01-27:
We do compress by default as we cannot know how big the data could be (could be an image). Compressing the data also help to know which kind of matrix it is (the matrix type and dimensions are encoded in the compressed byte array saved in the database).

We indeed use zlib. 
https://github.com/introlab/rtabmap/blob/c27507bc772cfda34f6624d386ab2cd62d49e90d/corelib/src/Compression.cpp#L247-L259

We have also a python example to uncompress:
https://github.com/introlab/rtabmap_ros/blob/master/rtabmap_python/src/rtabmap_python/compression.py