introlab/rtabmap #1206: UserData in Link Table

360wcui, 2024-01-25:
Hello,  I notice that there are some user_data in Link table of rtabmap.db.   What does the data in that column contain?   I am able to read the UserData from Data table, but why we have another user_data column in Link table?  Any insight would be greatly appreciated. 

[image]

360wcui, 2024-01-25:
Another note.  The size of user_data in the two tables is different.  
[image]

matlabbe, 2024-01-28:
If you are using laser scans, the `Link`'s `user_data` would be updated on proximity detection by default following this parameter:
```
Param: RGBD/ScanMatchingIdsSavedInLinks = "true"           [Save scan matching IDs from one-to-many proximity detection in link's user data.]
```
That is mostly used for debugging proximity detection and visualize in `rtabmap-databaseViewer` the scans used for it. The usage of `user_data` in `Link` is not exposed, in contrast to one in `Data`.

Another parameter than can affect `user_data` in `Data` table, though not enabled by default:
```
Param: RGBD/GoalsSavedInUserData = "false"                 [When a goal is received and processed with success, it is saved in user data of the location with this format: "GOAL:#".]
```
Examples of using `user_data`:
 * Saving WiFi signal strength: http://wiki.ros.org/rtabmap_ros/Tutorials/WifiSignalStrengthMappingUserDataUsage
 * Saving objects detected and there relative position from the base frame: https://github.com/introlab/rtabmap_ros/blob/7845e530959e09cc4f6d10d69cca766c9d1ff775/rtabmap_demos/src/SaveObjectsExample.cpp#L80-L95

closed 2024-01-28 by 360wcui